scottbez1/smartknob #13: Is USB required to operate?

nagyrobi, 2022-03-12:
Would it be possible to apply power from underneath, and not the USB-C port?

In a wall-mounted setup, having a USB plug sticking out would make it look dull. Instead, one could mount on top of a junction box within a wall, with a power supply underneath.

scottbez1, 2022-03-13:
There are 5V and GND testpoints on the bottom of the PCB you could use to power it without needing USB. This is a direct connection to the 5V/VBUS net (without any diode or other power switching protection) so you should take care not to plug in USB while also powered via the 5V test point or you may backfeed power on the USB port.

[image]

closed 2022-03-13 by scottbez1

nagyrobi, 2022-03-14:
Just a suggestion: 
- have prepared a location close to the center of the board for one to solder on a socket like 2EDGR-3.5-2P to the bottom of the board, so that it sticks towards the junction box in the wall
- have 4 holes in the pcb corners so one can simply screw it on the wall