scottbez1/smartknob #14: ESP with LAN with PoE

nagyrobi, 2022-03-12:
Is there any plan to have a version supporting wired Ethernet connetion? 

ESP32 can accomodate with various Ethernet PHY chips, also combined with an 802.3af PoE supply this project could really kick ass in the home automation world. Think of this also as mounted on wall on the end of a CAT5 cable supplying both data and power...

scottbez1, 2022-03-13:
No plans for Ethernet. It would definitely be cool, but probably not something I will personally prioritize at the moment. My main priorities are:
- better documentation
- firmware improvements (graphics, configuration api, serial protocol, getting wifi working)

But the other issue that comes up is limited IO - with the current ESP32 module I'm using, all outputs are currently in use (except IO0 I guess) and the only open pins are inputs (34, 35, and 39).

The LCD (6 GPIO) and motor driver (7 GPIO) are both IO heavy, and then there's I2C, magnetic sensor, and strain gauge amplifier, so it would require a pretty substantial change to be able to free up pins to support an ethernet PHY.

[image]

There's probably a hybrid approach where you use an external ethernet<->serial adapter and connect this in place of the CH340 USB serial interface, but I'm not familiar with what's available in that space, or how to integrate it. In that case it would be helpful to expose RX and TX as testpoints for easier connection.

nagyrobi, 2022-03-14:
I suggest for develompent on the longer term... maybe you could use a GPIO expander like MCP23017 for more basic stuff...

Ethernet-Serial conversion limits functionality a lot, as there's an extra middleware needed to convert serial-like protocol to whatever is needed. Ethernet connection should be equivalent to wifi in terms of functionality.

scottbez1, 2022-03-15:
Added RX/TX testpoints, so hardwired serial communication without a USB connection is easier to hook up if desired.

However, the remainder of the discussion in this issue regarding first-class ethernet support is outside the scope of planned features, so in the interest of keeping the issues list clean, closing this as wontfix.

scottbez1, 2022-04-07:
Per #51, I'm reopening this issue as a general place for brainstorming/discussion around ethernet and PoE. Like I mentioned in that issue, I want to keep this project extremely focused for now, so I don't see ethernet support realistically being added any time soon, but doesn't hurt to have a place (this thread) for discussion in the meantime

austinscreations, 2022-04-13:
for POE i've been using https://www.aliexpress.com/item/1005002768318128.html
(keyword: SDAPO DP1435 - should link die)

combined with the w5500 spi ethernet IC - which requires 5 pins. sck, miso, mosi, cs, and a reset pin (not mcu reset - needs to be controllable IO)

Timvrakas, 2022-09-15:
I'm not sure Ethernet is needed for this simple of a device, PoE adds a significant cost and complexity electrically. If you were integrating this into an industrial/home automation system I can see the appeal, but if I were running more than one or two units I think I would quickly prefer RS485 or CAN or BASE-T1L or similar to link them to a controller that in turn had Ethernet.

If you're looking for a PoE module, I will recommend https://silvertel.com/ they make high quality PoE modules.

sanderkooger, 2023-10-19:
Another option would maybe be skip the POE and Ethernet connectivity, Mount the smart knob on a (in wall) switch box and use the power thats already there? Then use WIFI for communication. 

Is this a viable option too?

sanderkooger, 2023-10-19:
@scottbez1 Congrats on this project BTW, This could be the start of something really cool. 

At the moment, I'm researching lights control for my home. If Wi-Fi becomes available and i could have the knob send out API requests over Wi-Fi. I should be golden. I'm also looking into thermostats that i can talk too over an API. 

Have you ever thought of promoting your project on the Wled https://github.com/Aircoookie/WLED

I think you would find a lot more enthusiasts there who have and electronic knowledge and could maybe contribute.