scottbez1/smartknob #16: Motor out of stock

mattkerrison, 2022-03-14:
Not a major issue as you've said you've got several others on the way for testing, but the link to the motor you used is now out of stock:

[AliExpress](https://www.aliexpress.com/item/1005001384349991.html)

scottbez1, 2022-03-14:
Yeah, I'm hoping this is just a temporary out of stock issue, but the seller actually has another listing that looks essentially identical (though a tiny bit more expensive) and is still in stock (for now) - https://www.aliexpress.com/item/4001356858801.html

I've ordered a handful of those myself and will confirm whether they're the same in ~2 weeks when they arrive :stuck_out_tongue: Probably worth snagging a few now regardless since they're pretty cheap.

My fear is that they may just be selling off "new old stock" from some short-lived product and the motors are not actively being manufactured any longer, in which case the interest in this project could quickly consume what's available. :crossed_fingers: Hopefully that's not the case though, as there are basically zero other cheap options for hollow shaft BLDC motors with low KV and zero cogging torque that I could find...

SpuriousGer, 2022-03-25:
Your update on March 24th seems quite discouraging, does this mean the project is dead? Seems like one of the coolest ones I've seen in ages.

C44Supra, 2022-03-25:
It gets worse, the other listing that @scottbez1 linked to is gone now too. I had an order for a couple of those and that order was simply cancelled earlier this morning, so that double listing seems to have been an oversight. The motors from that seller (and two other sellers as well) that looked similar by had wires running through the hollow shaft already, those have also been removed. Well not removed, I created two orders from separate sellers last night and both of those items have been changed into a completely different item. I haven't been able to find a replacement motor yet.

I'm not sure what other motors @scottbez1 has bought to test, but I would love to know. It would be very helpful to know which motors we should avoid specifically.

scottbez1, 2022-03-25:
Responding from my phone, so pardon the brevity.

I don't want to call the project dead quite so soon, but it's certainly not
looking very good at all...

Regarding other motors, I basically ordered every single small BLDC from
that AliExpress seller (which is most of the cheap motors I could find on
AliExpress in general) and there's not good news there, unfortunately. All
of the other hollow-shaft motors had noticeable or substantial cogging,
making them poor candidates.

I'll follow up with more details when I have some time this weekend,
including a list of all the motors I've personally checked.

C44Supra, 2022-03-25:
I certainly wouldn't call it dead in the water, just needs some adjustments for a more widely available motor. I'm tempted to grab a more common 2804 or maybe 2204. Lots of varieties though and not a whole lot that either have an axial magnet or room for one.

SpuriousGer, 2022-03-25:
> Responding from my phone, so pardon the brevity. I don't want to call the project dead quite so soon, but it's certainly not looking very good at all... Regarding other motors, I basically ordered [...]

You mention cheap, is it a prerequisite that it is really cheap or are there other motors that might fit the bill but are more expensive?

C44Supra, 2022-03-25:
@scottbez1 would be helpful to know what sort of BLDC motor you're using at the moment. From what I understand the majority of cogging is a result of the type and size of magnets used in the rotating part. In an effort to generate as much torque as possible, most manufacturers opt for neodymium magnets which are significantly more powerful compared to ferrite magnets.

jimius, 2022-03-26:
I don't know a lot about motors, but I did find this:
https://store.cubemars.com/ from a company called Tmotors (
https://uav-en.tmotor.com/).
Again, no idea if the specs match anything you need.

dsamarin1, 2022-03-26:
Seconding what @C44Supra said, perhaps logging here the specs of your currently best-performing motor so the group can help hunt down a replacement, especially once the AliExpress pages disappear. I'm thinking these are important: Critical dimensions, Kv, electrical specs (ex. maybe max torque can indicate something else useful), keywords. Others?

In the future, the move is probably a redesign for the more common outrunner motors on hobby RC websites. Considering tackling part of this myself.

LeonBentrup, 2022-03-27:
It looks like [newegg](https://www.newegg.com/p/298-00HJ-001V2) has the motor but only with a slipring installed. removing that wouldnt be a problem, but the 50$ pricetag is. i think the best way going forward would be using a standard gimbal motor and putting on a diametrical ringmagnet from [mcmastercarr](https://www.mcmaster.com/ring-magnets/direction-of-magnetization~through-diameter/) or so. something like those 2804, 2805or 3413 look more common and could propably be used

C44Supra, 2022-03-27:
@LeonBentrup that listing on Newegg is from a chinese reseller as well. It was listed by at least 3 different vendors on AliExpress but all three have removed the exact same listings at the same time. I'd say there's good chance it's one and the same vendor, just active under a couple of different names. That said, I did get a reply from the AliExpress seller earlier today:

> Sorry to tell you that we can't send this item out. A local buyer picked all this item a few days ago and our workmate forget to remind us to end this listing. 
We contacted several suppliers and they don't have this item in stock. And also can't give sure reply when the item will be available. 
But we have the other similar item in our shop,could you please check it meet your need or not?

The other item they are referring to is https://www.aliexpress.com/item/1005003751799249.html It looks somewhat similar but is missing the diametrical magnet. There's also not enough shaft protruding to stick a magnet onto so that would be a no good either.

It's also quite problematic that their motor numbers seem to reference the outer diameter and height of the entire motor, where as other suppliers generally reference the internal dimensions of the actual motor parts. The 3413 for instance is pretty much the same as a 2804.

I did "some" looking around on AliBaba last night / this morning ( / what day is it again?), and found a supplier that has a motor that shows potential. They're not too expensive, have a 7mm diameter protruding shaft on the bottom and looks to be good design. I've contacted them and are waiting for a reply with more data about that specific motor.

A quick search on Amazon led me to some 10x3x7 (outer diameter x thickness x inner diameter) diametric magnets. I think thinner ones would work, too, though.

LeonBentrup, 2022-03-27:
@C44Supra Does it have to to be a motor with a stick out shaft? for example the mks servo42 just uses superglue and two of those are running fine on my 3d printer. its not a hollow shaft there but those [cheap 2804](https://de.aliexpress.com/item/1005001968864724.html?spm=a2g0o.detail.1000014.7.18a711a1IIFzjm&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40000.267768.0&scm_id=1007.40000.267768.0&scm-url=1007.40000.267768.0&pvid=cdc11d7c-d8bd-49ea-8f4c-22b666c9050f&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40000.267768.0,pvid:cdc11d7c-d8bd-49ea-8f4c-22b666c9050f,tpp_buckets:668%232846%238112%23502&pdp_ext_f=%257B%2522sku_id%2522%253A%252212000018267338478%2522%252C%2522sceneId%2522%253A%252230050%2522%257D&pdp_pi=-1%253B3.03%253B-1%253B-1%2540salePrice%253BEUR%253Brecommend-recommend) look like you could glue it on there directly. Sticking a 6mm rod with some paper(to not glue the rod in there) down the hollow shaft you could even get it centered pretty good. and since its not turning with 2000 rpm i wouldnt be afraid of the glue giving up

C44Supra, 2022-03-27:
@LeonBentrup it'd have to stick out of the bearing at least a little bit, otherwise there's nothing to glue onto without interfering with the motor. I was considering that too. The motor you linked to is a 20-24V BLDC motor though, I haven't dug into the schematics too far just yet, but I suspect the motor in the current design is driven at 5V, which would likely not be very impressive to a 24V motor. The motor I linked to earlier would probably be a safer bet (a 3413), but I'm somewhat hesitant to place an order with that vendor again, considering it took him 10 days to just cancel my order with zero feedback.

LeonBentrup, 2022-03-27:
@C44Supra the linked motor was just an example but thanks to usb c even 20V isnt a problem. the pcbs look quit empty so there should be enough space for a ip2721 or ap33771 to get what ever usb-c pd voltage you want just with the IC and some resistors and buck it down to the 3.3v the esp32 needs. 

And looking at the linked motor you could propably get away with glueing a 8x6x3 magnet directly on to it. but you could also print a little spacer and glue it in to get the stickout. or use a off the shelf 6x0.5mm steel pipe. this would propably be the simplest solution assuming you dont need the full 6mm space in there. even the 5mm should be enough for spi,power, rst and backlight.

C44Supra, 2022-03-27:
@LeonBentrup USB-C could indeed provide higher voltages. However the TMC6300 driver only does up to 11V so you'd likely have to account for that too. Not too hard, but something to keep in mind.

LeonBentrup, 2022-03-27:
@C44Supra And there could be a problem. at the moment you cant really get any normal to high power BLDC ICs. i looked at most of the common ones for other projects. But with USB-C PD you could also get 9V which is also the lowest voltage a 3S LiPo should get. so BLDCs that work with 3S batterys should also work with that. that also makes it possibile to still use a linear regulator cause it only needs to drop 5.7V*I and not 16.7V*I when using 20V which is basicly impossible without a heatsink. 

For motors that would mean you could use a standard t-motor GB2208 but those are pretty thick but propably have a good availability. or something like those [MiTooT](https://de.aliexpress.com/item/32953126469.html?spm=a2g0o.detail.1000014.30.607e580bNTltCL&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40000.267768.0&scm_id=1007.40000.267768.0&scm-url=1007.40000.267768.0&pvid=7d753a74-acbb-465e-96b0-08dd5378524b&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40000.267768.0,pvid:7d753a74-acbb-465e-96b0-08dd5378524b,tpp_buckets:668%232846%238112%23502&pdp_ext_f=%257B%2522sku_id%2522%253A%252267386558847%2522%252C%2522sceneId%2522%253A%252230050%2522%257D&pdp_pi=-1%253B16.83%253B-1%253B-1%2540salePrice%253BEUR%253Brecommend-recommend)/[Flycat](https://de.aliexpress.com/item/32929665214.html?spm=a2g0o.productlist.0.0.2ac1586aYyD9KI&algo_pvid=bad9764c-c835-404f-bf17-76185e12a7a0&algo_exp_id=bad9764c-c835-404f-bf17-76185e12a7a0-8&pdp_ext_f=%7B%22sku_id%22%3A%2210000001579149334%22%7D&pdp_pi=-1%3B16.67%3B-1%3B-1%40salePrice%3BEUR%3Bsearch-mainSearch)/whatEverRandomName 2805 motors. 
I also found this ["Lidar" motor](https://de.aliexpress.com/item/1005004036821327.html?spm=a2g0o.productlist.0.0.2ac1586aYyD9KI&algo_pvid=bad9764c-c835-404f-bf17-76185e12a7a0&algo_exp_id=bad9764c-c835-404f-bf17-76185e12a7a0-30&pdp_ext_f=%7B%22sku_id%22%3A%2212000027817000823%22%7D&pdp_pi=-1%3B17.66%3B-1%3B-1%40salePrice%3BEUR%3Bsearch-mainSearch) but this also looks proprietary so there is a chance it coudl be out of stock after a few hundred sales again.

MarcSN311, 2022-03-27:
3D printed BLDC motors exist. Could we design the perfect one for this project?

gcormier, 2022-03-27:
I shot a message to Carl at Flexar to see if there's any opportunities here.

scottbez1, 2022-03-28:
Some specs and photo from the inside of the original motor in case it's helpful identifying an alternative:

[image]

(for reference, the diametric magnet is a ceramic ring and appears to have been glued onto the rotor, so leverage is not the best way to remove it in one piece... :grimacing: but it's necessary to remove to access the retaining ring)

- 32mm OD, 5.9mm ID
- 14 poles (7 pole-pairs)
- 12 slots
- U-V coil resistance of ~8 ohms
- essentially zero cogging torque (this is a very rare attribute from testing a bunch of small "gimbal" motors)
- according to original product listing, 3280RPM at 12V, or 273KV (this is untested, but does match the expectation for a gimbal motor to have a low KV rating)

dsamarin1, 2022-03-28:
@scottbez1 Ok this is starting to make sense. Those look like compressed ferrite material, which will have a lower flux density than the little bar neodymium magnets, and so lower max torque output but also lower cogging torque. I think.

scottbez1, 2022-03-28:
Other motors I tested previously and decided against:

[image]

### On the left:
- 34.5mm OD, 4.9mm ID
- U-V resistance somewhere around 30 ohms
- too much cogging torque
- poor internal construction - solder or wire was rubbing against rotor internally

[image]

### On the right:
- 34mm OD, 6mm ID
- U-V resistance ~9.5 ohms
- too much cogging torque

[image]

But even if those motors did fit the bill, I think they still wouldn't be a good option since they'd likely go out of stock immediately as well. I think the challenge will be to find a good motor (hollow shaft, low cogging, low kv) that's still actively manufactured rather than leftover "new old stock" like these.

C44Supra, 2022-03-28:
That would indeed be a ferrite magnet in the rotor. Usually manufacturers put neodymium magnets in instead to increase motor torque at the cost of top speed. I'd say it's probably hard to determine which type of magnet a motor contains if you don't have a clear visual of it or take it apart.

The last motor you added a screenshot of lists the neodymium magnet type in the title at least (NdFeB). That said, even if it was a ferrite magnet I wouldn't trust that particular seller to deliver anything. Was the board as is able to drive the 20-24V motor at all?

In other news, the manufacturer I've been in touch with has only sent me dimensional drawings so far. Still waiting to hear back what sort of magnets and arrangement there is in the motor.

18Markus1984, 2022-03-28:
Did someone wrote the manufacture of the original link for the motor why all these listings got taken down?

C44Supra, 2022-03-28:
@18Markus1984 see one of my earlier replies. They claim that a local buyer bought out their entire lot of motors and someone just forgot to pull down the listing.

dsamarin1, 2022-03-28:
> I'd say it's probably hard to determine which type of magnet a motor contains if you don't have a clear visual of it or take it apart.
@C44Supra Could a starting point for trying to identify motors with ferrite magnetics be to look for ones of a certain size that have a significantly lower maximum torque than others? Just applying this as a pre-filter to try and narrow down on options quicker.

C44Supra, 2022-03-28:
Possibly. The original motor listing didn't have a torque spec, so no reference there. But you can obviously compare different motors to each other, significantly lower torque ratings could mean ferrite magnets. Or typos. Or intentional misrepresentation. AliExpress / eBay and genuine parts specifications generally don't go hand in hand unfortunately 😆

stanlee-1024, 2022-03-29:
> @18Markus1984 see one of my earlier replies. They claim that a local buyer bought out their entire lot of motors and someone just forgot to pull down the listing.

Hi, do you have any contact information of the seller? It will be easier for me to talk to them from here China.

C44Supra, 2022-03-29:
@stanlee-1024 The seller that had the original motors used for the project? That seller can be found via the link in 2nd reply in this thread. (or just this link: https://www.aliexpress.com/item/4001356858801.html). I wouldn't waste my time on that seller to be honest, they simply resell overstock items it seems.

The manufacturer I'm talking with hasn't provided anything tangible yet either, I'm still assessing what they're actually offering. Once I know more I'll share, but in the meantime I'll reach out to a couple of other manufacturers.

flambert860, 2022-03-29:
What about this one, the KV is lower, but many of the other specs seems right, perhabs search for gimbal replacement motors, they might be more reliable to maintain a stock.
https://www.alibaba.com/product-detail/Flashhobby-GBM2606-90T-brushless-DC-motor_1600077006273.html?spm=a2700.galleryofferlist.normal_offer.d_title.5c6f2b85BJtKN2

C44Supra, 2022-03-29:
Lower KV shouldn't really be much of an issue as the project doesn't need the knob to turn at high speed by itself. Even then, at 120KV it'd still be capable of roughly 600RPM at 5V. This particular motor only has a 4mm shaft though, not sure how they stuck the magnet on the shaft.

That said, the torque seems a bit high at "450 grams", that could indicate neodymium magnets and therefor excessive cogging.

C44Supra, 2022-03-30:
There is a more widely available GBM2804H-100T, which is slightly larger in diameter but has a 6.5mm shaft bore. No clue what sort of magnets either have though, I will be trying to get in touch with some suppliers to see if they can provide any additional information beyond what is listed in the product listings.

scottbez1, 2022-03-30:
Let us know who you talk to and what you find out from them @C44Supra (hopefully we can avoid having everyone spamming the same suppliers and asking all the same questions, so thanks for taking the lead on that)!

I will reach out to the supplier @flambert860 linked to get some more info and potentially order samples from them, but if there are any other motors that seem promising let me know (either post to this issue or email me directly at smartknob@bezeklabs.com) and I would be happy to order a bunch of different samples and share any findings.

mzhboy, 2022-03-30:
How about [HT2804/HT2806](https://www.aliexpress.com/item/10000203187344.html) . $10 without encoder on taobao

> Seal enclosure to prevent dust and dirt contamination
Hollow shafts are used for slipring or wiring
Specifications:
Copper wire (OD) 0.21mm, prewound 60 times.
interface resistance is 5.1 ohm
Motor size: 34.6mm (OD) * 18.3mm (height)
Stator dimensions 28 mm (OD) x 4 mm (height)
Hollow shaft diameter 4.5 mm
configuration 12 n14p
camera weighs 100-300 grams

[image]
[image]
[image]

C44Supra, 2022-03-30:
Just heard back from one manufacturer. Apparently the C2804A / C2804B / E2804 (and similar ones) all use neodymium magnets. They're looking into options for ferrite magnets, but that's likely going to be a custom production with a custom price tag (in other words: expensive).

jamesgao, 2022-03-30:
Would it be possible to actively drive the motor out of the cogging detents using the encoder inputs? Perhaps adding "viscosity" to the feel of the knob would also be an option?

flambert860, 2022-03-30:
I think the best solution would be to find a popurlar gimbal where the motor fit the specs, then there should be plenty of replacement parts

scottbez1, 2022-03-30:
I'm waiting on sample pricing for the GBM2606 (though I'm worried that the smaller ID may be too small to accommodate a 3d-printed riser) and GBM2608 (wide enough ID but overall taller), along with a quote for a small initial batch. It sounds like they can make them with either individual neodymium magnets or a "magnet ring" for less cogging, so I'm inquiring about getting them made with the ferrite magnet ring.

There would still be the issue of adding a diametric magnet, but one possibility would be to mount it onto the top of the rotor and move the MT6701 sensor onto the screen PCB instead of the base PCB. The magnet could be inset into the rotor spacer (it's 1.8mm thick on top, IIRC) to avoid increasing the height too much.

Or a potential alternative would be using 2 thin circular magnets (like 4x1mm) with one facing up and the other facing down to create a diametric magnetic field slightly above them, like so:

[image]

(that's from an earlier prototype before I found the 3215 motor with built-in diametric ring)

jtsfour, 2022-03-31:
Has anyone looked at the JD-Power Motor manufacturer?  This motor looks pretty close http://www.jdpowersky.com/en/p-info.aspx?cid=1&id=46  and this one http://www.jdpowersky.com/en/p-info.aspx?cid=1&id=49  They have an array of models and sizes.

They have an Alibaba store located here https://jdpowersky.en.alibaba.com/productgrouplist-803362553/Flat_outrunner_gimbal_Series.html?spm=a2700.shop_plgr.98.3  Here is one of the listings I was looking at.  https://www.alibaba.com/product-detail/JD-power-DC-3514C-2804-hollow_60489120691.html?spm=a2700.shop_plgr.41413.37.95ae517fSFGbWm

I have been loosely following the project so I am not sure if I missed something about these that is significant or would not work.  It looks like the DC-3514C is 1mm larger in diameter. ~6.5mm shaft. 16mm thick. (dimension diagram in first link.)  What do you guys think?

mzhboy, 2022-03-31:
[PM3505](https://www.aliexpress.com/item/33027092449.html) has buildin sensor and 8ch slip ring . But it's expensive ￥108 CNY without sensor

[image]

jtsfour, 2022-03-31:
I have been taking a look at HDD motors.  Every image of the inside of an HDD motor that I have seen uses the same (or a similar) magnet material that is inside of the original motor design.
[image]

It may be possible to drill out a hollow shaft from the motor.  However, most HDD motors are quite wide and there are many many designs.  Its just an idea I thought I would throw out there.  Motors from a 2.5" HDD might work but I don't know if they can make enough torque to be useful.  I will take apart a 2.5" drive later and see how big the motor is compared to the original motor shown above.

C44Supra, 2022-04-01:
@PixelGalaxxyy 
The first section of the model number is the item number of the original motor on AliExpress. The rest of the numbers, combined or in chunks don't show up on google in any relevant way.

I'd say it's very likely that this is either the same seller, or this seller bought their stock from the seller that the original motor for this project was bought from.

@jtsfour I've been in touch with a number of vendors on AliBaba over the past few days. So far, every single 2804 motor has neodymium magnets. The 2804 style motor is being sold in a variety of configurations (4mm hollow shaft, 6.5mm hollow shaft, solid shaft, with and without shaft protruding from the bottom, etc.). The "Changzhou Belmont Intelligent Motion Co., Ltd" representative told me that all their motors (anything starting with BM and then a model number) have neodymium magnets and are to be considered finished designs where ferrite rings are not a customization option. The same representative works for "Changzhou Mcmaster Electrics Co., Ltd" as well. Considering their motor model number start with BMM, it's very likely just a proxy.

I'm still waiting for a response from "Shenzhen Two Ant Technology Co., Ltd.". Supposedly they have a ferrite ring magnet motor, but I'm waiting for a photo of the rotor and said magnet.

@mzhboy that is another one of the motors I've been looking at. If I remember correctly, a visually similar motor is available from a number of manufacturers on AliBaba, with and without the encoder. All of them tend to be up there in terms of price. I'll engage with a few of them tomorrow to see what I can dig up.

@jtsfour @dsamarin1 HDD motors are probably the easiest source of low cogging motors. All the disks I have dissected over the past few years have had integrated motors, where the stator is part of the lower part of the aluminum housing. If you can find one small enough and can get it removed from the remainder of the housing you could very likely use it for this project. However, these motors have very low torque, which is good for cogging, but bad for the actual feedback loop. Plus you'd still have to modify the shaft and a good chunk of the plastic components of this project.

jtsfour, 2022-04-01:
@C44Supra 

Thanks for the input.  I was just now doing some digging into the magnets.  What if we standardized around a motor design even with neodymium magnets (Like the 2804).  Couldn't we just remove the Neodymium magnets from the rotor and replace it with a Ferrite Ring of our own specifications?  How are these magnets installed?  Could we use a heat gun and pull them?

I found this magnet supplier that does custom multipolar ferrite rings.  Maybe this could be an easier option over finding motors that have been specifically built weaker and cheaper.  We could use a small crowdsource and buy a couple thousand and use them collectively.

https://www.hsmagnets.com/product/customized-radial-multipole-ferrite-rotor-magnets/

I am not an engineer.  Maybe someone with more experience could weigh in on this thought?

C44Supra, 2022-04-01:
@jtsfour

That would certainly be an option. I don't have any motors on hand just yet, but I suspect the magnets are being glued in, likely with a two part epoxy of sorts. The likelyhood of removing the magnets without destroying them or the rotor itself are likely slim to none.

That said, you may be on to something. I'll ask the manufacturers if they'd be willing to sell motors without magnets in the rotor at all.

Multipolar ring magnets can be found in all shapes and sizes on Alibaba as well. Finding one that fits the bill might be a bit of a challenge though.

mcmadhatter, 2022-04-01:
I have a "Power Motor GBM2804R 2804 Gimbal Motor with Aluminum case /Slipring /AS5048A Magnetic Encoder for camera stabilizer" on order for something else, but I'll try it out and see what it is like beforehand.

scottbez1, 2022-04-01:
Just for the heck of it, I submitted an Alibaba RFQ with the details and photos of the existing motors (32mm rotor, 6mm ID, 14P12N, etc), just in case the original manufacturer is still around and can easily spin up production of it again, or in case another manufacturer is able to produce an identical product. Certainly would be nice if it can be found and avoid the need to redesign the mechanics. Probably unlikely, but figured it's worth a shot.

scottbez1, 2022-04-02:
I've ordered some samples of the 2608 with ring magnet (from "Flash Hobby" Huizhou Feishun Technology), asking for a couple 5 ohm and a couple with 8 ohm windings to see how they compare.

Apparently there weren't readily available ferrite ring magnets for the 2606 size, so they will not make samples of that size for now.

Samples will ship via DHL Express, so depending on how long manufacturing takes I would hope to have them within a few weeks to evaluate them.

A big thank you to all the GitHub [sponsors](https://github.com/sponsors/scottbez1/) for helping fund experimentation like this! :heart:

If you're interested in eventually purchasing a motor (still a long way away and motor still TBD) - can you fill out [this quick anonymous survey](https://forms.gle/PtuVDrpcA2niVDCL9) so I can get a sense of total interest quantity, geographical distribution, and pricing? To somewhat set expectations, based on what I'm seeing I would guess that it will cost somewhere between US$25-50 per motor unless we can find something suitable that's already being widely manufactured and distributed, unfortunately...

 And of course let me know if you find other interesting motors or suppliers and I am happy to place more sample orders!

scottbez1, 2022-04-02:
@jtsfour thanks for pointing out DC-3514C earlier. The JD-power contact just reached out to me from my RFQ mentioning that DC-3514C and DC-3515C are both similar to what I was looking for. It looks like DC-3515C has a "ring magnet", so I'm looking into ordering samples of that one - they also confirmed it has lower cogging than 3514C.

[image]

(oddly they show the 3515C with a non-hollow shaft and a gear/sprocket on their website, but it sounds like it is available with a hollow shaft as well)

They are offering the option to add the diametric magnet to the bottom, so I'm planning to order a couple samples with the magnet and a couple without. They will be 5.4ohm coils. I will update this post as I get more info about these motors/samples.

---

**Reposting for continued visibility:** If you are interested in purchasing motors, help me understand general interest/quantities by filling out this [quick motor survey](https://forms.gle/PtuVDrpcA2niVDCL9) (and thanks to the 27 of you who have already filled it out!).

jschoch, 2022-04-02:
you may be able to do anti-cogging in the software.  there are some threads on the simpleFOC forum about how to do this.  I think any "gimbal" type motor would work well.  My original test was using a 2204 motor, but I found an amazing deal on 10 used BGM5208-200 motors.  They both worked well with simpleFOC.  If you really really want true cogless you should search for "slotless" however I didn't see any hollow shaft slotless motors on ali.  the more poles the better so that may be another parameter to search on.

Over2247, 2022-04-02:
I've found an interesting website of a reseller.
They can produce customized motors in bulk, for cheap. 
Maybe they'll go as far es producing motors with ferrite magnet rings instead of neodym ... 
I'll drop the link, maybe someone wants to look into that ¯\_(ツ)_/¯

https://www.starselectron.com/se-bldc-e2804-ptz-motor-camera-motor/

modi12jin, 2022-04-03:
[image]

https://www.made-in-china.com/

@scottbez1 I'm not too sure what motor is needed, maybe this site helps.

jonny772, 2022-04-03:
How about a spotless & frameless motor? Loads of space then for the printed shaft. I need to have a looks around for financially viable options

 https://www.thingap.com/standard-products-slotless-frameless-brushless-bldc-high-torque-to-weight-ratio-large-through-hole-motor/

jorisplusplus, 2022-04-03:
Has anyone already looked at this motor? https://www.aliexpress.com/item/1005003645403776.html
Looks very similar construction, only problem I see is the 240 degree rotation limit.

jonny772, 2022-04-03:
The original motor is also in stock here, it's another distributor so it might not be in stock for long. I've ordered one and will report back if it actually arrives. 

https://gbr.grandado.com/products/micro-mini-32mm-hollow-shaft-brushless-ptz-motor-aperture-6mm-for-small-uav-brushless-gimbal-motor-handheld-camera-gimbal?gclid=CjwKCAjwi6WSBhA-EiwA6Niok49qBVjhUWv4ciKbsTjD6yXHCEdTRbQOHq_k5nJFHh707kAINS8joxoCepIQAvD_BwE&variant=UHJvZHVjdFZhcmlhbnQ6NjM0MjIxNzg

ewingate, 2022-04-03:
> Has anyone already looked at this motor? https://www.aliexpress.com/item/1005003645403776.html Looks very similar construction, only problem I see is the 240 degree rotation limit.

It looks like they've done a lot of damage to get the, "what's inside" photos. If it can be opened without damage that rotation stop should be easy to deal with.

larryqiann, 2022-04-04:
> Has anyone already looked at this motor? https://www.aliexpress.com/item/1005003645403776.html Looks very similar construction, only problem I see is the 240 degree rotation limit.

I have purchased from this "Esinou Store" before. I had a positive experience with them, but they appear to mostly sell overstock/other misc motors, so I would imagine continued interest will deplete the stock.

tascord, 2022-04-04:
Maybe a good idea to start up a compatibility table with product/part numbers?
That way people can add motors they know work along with a video in the pr to make sure its fully functional? (Run through a suite of tests)

scottbez1, 2022-04-04:
@tascord yeah, we can make a compatibility table once some compatible motors are found... that remains the challenge at the moment...

The VAST majority of readily available BLDC gimbal motors exhibit cogging, which prevents smooth rotation and limits the usefulness of software-defined detents since you're fighting those hardware magnetic detents. (There's maybe a possibility of applying anti-cogging in software as @jschoch mentioned, though most of what I've seen has been cogging compensation for a *driven* motor rather than an *externally actuated* motor that is sitting "idle", and it would also greatly complicate the code and calibration procedure).

Here's a photo of the cheap BLDC motors from AliExpress I've tested and have confirmed **all** of these exhibit cogging (not all of them are hollow core, as I was also looking for motors to make a screenless haptic dial)

[image]

So for now the hunt continues with zero compatible motors confirmed so far, and it is still looking like we may have to get some motors custom manufactured for this. Most of the sub-US$10 motors on AliExpress etc are likely to be overstock rather than actively manufactured, so even if one of those was confirmed to work, it's very likely it would go out of stock nearly instantly like the original motor.

---

**Reposting for continued visibility:** If you are interested in purchasing motors, help me understand general interest/quantities by filling out this [quick motor survey](https://forms.gle/PtuVDrpcA2niVDCL9) (and thanks to the **146** of you across *30 countries* who have already filled it out!).

larryqiann, 2022-04-04:
I'm wondering if anyone has looked into whether CD drive motors or motors for the mirror in laser printers would work, or whether some manufacturer could produce variations on one for this. The ones I have seen produce close to nil cogging, and look like the construction is much simpler than the gimbal motors.

dsamarin1, 2022-04-04:
@scottbez1 Any commonalities in those motors that you've detected? I'm thinking about number of poles/winding configuration, magnet dimensions, DC resistance.....

paulsvc, 2022-04-05:
Hello
I'm following the project since the YouTube video upload.
I'm more a machinist than electronic engineer or whatever.
Taking into account all efforts you make to find a supplier and the product seems to not be standard ( so it will always be stopped one day or another). As it has already been suggested maybe put the same amount of effort into a 3d printed model with an open source design.
As you do not required high speed or high torque, I assume this would not be complicated to design.
You can select the magnet you want, number of windings and wind size.

I hope this project will be available because I have a lot of use for this.

[365 more replies]