scottbez1/smartknob #17: Integrate with ESPHome

nagyrobi, 2022-03-14:
A quick suggestion: ESPHome is an exceptionally well though-out modular environment where most of the work has been done in terms of integration with many third party systems. If you would develop your I/O as an ESPHome component, you could save tons of work and concentrate on your special features supported by the hardware.

Also have a peak at OpenHASP on how LVGL is approached. It hardly relies on touch functionality, though, so may not be appropriate here.

cupertinomiranda, 2022-04-03:
Would love to work on the integration with ESPHome once I can get my hand on the required parts and boards.
Do you have boards available?

Didel, 2022-04-14:
Depending on the final use, [QMK](http://qmk.fm) might also be an interesting ecosystem to look in to. What ESPHome is to home automation, QMK is for keyboards: it’s an open source “keyboard” firmware with a lot of possibilities (rotary encoders are already supported). It would allow end users to configure the device however it would suit them and make use of QMK’s existing infrastructure (website, tooling, documentation, online configurator, … )

juli303, 2022-08-20:
I like the idea of creating a separate repo for a light weight firmware that targets your smart knob as a reference design but would also be open to integrate other hardware designs too. Kinda like OpenHASP but specifically for small (round) screen devices with a rotary input.
Support for HomeAssistant and other Integrations would probably be bigger that way, if it can target other hardware designs too.

The UI should make it easy to control a single device so you don't have to go through lists to turn on your lights but should also have the ability to control multiple things in a sleek way (eg. main function is turning on/off and dimming the lights in a room but you can also control the thermostat without going through tons of menus)

Notifications that can be shown on the screen and incoming events would also be a nice feature. I guess sending the light sensor data to home assistant would also be useful for some people.

Also a settings menu would be nice to setup wifi from the device itself and other stuff.

A big factor to keep in mind is the limited input for such a device. Right now we only have rotary input and pushing on the screen which could be further split down into single clicks, double clicks, long clicks and force sensitive clicks.

To not make this project easy to develop and use, a few basic UI and input components should be defined and reused instead of just building a spaghetti UI.

Would anybody else like to share their thoughts/opinion?

tango2590, 2022-08-29:
I'd absolutely love it if it could work with ESPHome. I've been sitting on this project waiting for hardware sources and updates to sizing to get sorted out so I can integrate it into my HomeAssistant instance. I'd be down for testing it too.

willumpie82, 2022-09-15:
I think that ESPhome integration might get tricky, when making a simple blank ESPhome project the footprint is already ~40% of the memory. if we start adding custom integrations will fill up the memory quickly. the current firmware size of Smartknob is already large. I would than suggest to move to MQTT in HA formatting that enables auto discovery, but already need optimizations (as mentioned in an other thread)

landonr, 2022-09-15:
I have an esphome firmware that works with a rotary here https://github.com/landonr/esphome-remote
I think it would be cool to get these rotary parts then I could build a similar UI to this project using the same backend code I'm using

Memodota, 2023-11-04:
So, if the limit is memory, may be change esp to something else?