scottbez1/smartknob #20: Screen PCB thickness

Zomhad, 2022-03-15:
First of all thanks for open-sourcing this awesome design! I just noticed that the readme states the screen PCB must be 1.2mm thick ("Ordering notes: Must be 1.2mm thick (not "standard" 1.6mm) per mechanical design.") but on the silkscreen of the PCB there's a big "0.6mm thick" label. Probably just a small copy and paste mistake from the ordering notes of the base PCB, thought I'd mention it anyway. :-)

scottbez1, 2022-03-15:
Thanks for bringing this up!

Both are actually correct, if you can believe it. 1.2mm thickness PCB is correct, and the 0.6mm thick label on the silkscreen is meant to go with the "VHB 5925" text to clarify that the expectation is for 0.6mm VHB tape as opposed to the more common VHB 5952 (isn't that confusing product numbering?!) which is 1.1mm thick.

Let me think about how to make this more clear though, as I can totally understand the mixup.

I've been thinking about adding virtual footprints to the screen PCB for the double-sided tape, in which case the silkscreen clarification is probably not needed anymore since I can put a part number and clarifying comment for the VHB items which will appear in the auto-generated BOM directly.

scottbez1, 2022-03-15:
(another possible option I'm considering is just switching to 1.1mm VHB 5952 since it's more commonly found anyway, and just add another .5mm of vertical clearance inside the knob to accommodate)

closed 2022-03-28 by scottbez1