scottbez1/smartknob #23: Pick & Place File

tallAldi, 2022-03-18:
Would it be possible to include a simple pick & place file in addition to the bom?
I'm not that experienced with smd soldering and would like to order the pcbs already assembled at factory, at least for the general stuff like resistors and capacitors.

scottbez1, 2022-03-18:
Are you thinking JLCPCB or another manufacturer? In theory this shouldn't be too difficult to support since I'm already using KiKit to export the gerbers (I may have to swap some LCSC part numbers for ones in JLC's PCBA part library), but one potential concern is that the finest pitch/most difficult part, the TMC6300, is not in JLC's part library for assembly so you'd still need to solder it yourself. At that point I'm not sure how much it helps to have the resistors and capacitors pre-assembled...

Here's a photo for reference - the small capacitors here are 0603 and the bigger capacitor is 0805:
[image]

tallAldi, 2022-03-25:
Yes exactly I was thinking JLCPCB. I think I can adjust the part numbers myself, the important thing to have is the coordinates in the picker files. 
If I want to use a different manufacturer I have to adjust the part numbers anyways.
Good point regarding the TMC6300 though, maybe I have to solder that one myself, but then at least I don't have to worry about the rest.

KartoffelToby, 2022-03-30:
maybe pcbway is an option with a https://www.pcbway.com/project/sponsor/ sounds good, looks like its possible to make a "include all" package here with 3D Print and PCB assembled

tallAldi, 2022-04-08:
thanks for the BOM and CPL but I think there is something wrong with your export scripts as the CSVs are not formatted correctly:
`<Error>
<Code>AccessDenied</Code>
<Message>Access Denied</Message>
<RequestId>04005S033CFPDBGC</RequestId>
<HostId>
hrsQX0Zd4Dm7yARZxQnLXX2OTdabTdvb/b2LqfneL4g2Y6e8PpY6g72ds2pVmiQwefNG6p+Uq7I=
</HostId>
</Error>`

D-Buckingham, 2022-06-14:
Dear Scottbez1,
I really enjoy your design and the implementation you did is simply awesome.
I am at the same point as tallAldi and I would really appreciate it, if you could add centroid files in the releases. I tried to extract it myself with altrium from the gerbers, but somewhere I am stuck between errors or incomplete lists...
Thanks a lot!!!
PS: I get the same error message as tallAldi.
PPS: My chosen manufacturer is PCBWay ;)

SpuriousGer, 2022-10-07:
So, I contacted PCBWay and they said they'd be able to support this. What I don't really understand is: what do I need beyond it. I need to upload something here: https://www.pcbway.com/quotesmt.aspx
But what do I fill out and which files do I need to upload?

kquinsland, 2022-10-08:
> What I don't really understand is: what do I need beyond it. I need to upload something here: https://www.pcbway.com/quotesmt.aspx
> But what do I fill out and which files do I need to upload?

They're looking for several files but at a high level, the data they need fits into one of three big groups:

- the grbl files; one for each layer of the PCB. All/Most of these layers should share a common origin or fiducial point and some layers may contain notes about a particular feature or intention. This is typically sent as a ZIP file and it's the bare minimum you need to get a PCB made.
- the list of part numbers. If you're not going to use pcbway parts, there's some additional glue around which parts you supply and which they do. I don't know what this glue looks like, but i'd bet $ that it involves some back/forth with their support staff.
- a file that indicates the X/Y location (relative to the common fiducial / origin) for each of the part number(s).

The third group is commonly called the "pick and place" file... which is being discussed here :D.

The exact format for each will depend on the place doing the work; they all have their own rules around what they can/can't do ... etc.

Google will help you locate the rules / supported features for a board shop and then the rest is figuring out how to get the design software to spit out the files in the format(s) that the board house will accept.

**right now** one of the bigger questions is around _where_ to get the boards made. Not every board shop has the right part(s) in stock. Ideally, there would be a single part supplier that has a close relationship with the board shop so we only needed one set of part numbers.

SpuriousGer, 2022-10-08:
> **right now** one of the bigger questions is around _where_ to get the boards made. Not every board shop has the right part(s) in stock. Ideally, there would be a single part supplier that has a close relationship with the board shop so we only needed one set of part numbers.

Thanks a lot for the thorough reply, very helpful!
I think we have a list of those pick and place files, don't we:
https://smartknob-artifacts.s3.us-west-1.amazonaws.com/master/electronics/view_base-ibom.html
https://smartknob-artifacts.s3.us-west-1.amazonaws.com/master/electronics/view_screen-ibom.html

Or is this again something else?

I am trying to work with PCBWay on how to get this done. But I am so limited in my skills, I basically have two things: money (so I am happy to pay anyone for their time to assemble those lists) and the willingness to engage with them to have prototypes build and take the risk.

kquinsland, 2022-10-08:
> Or is this again something else?

That's a "human friendly" version of the data.  Look at the `LCSC` column on either and compare with the `Mouser` and `Digikey`. No one supplier has a part number for everything that's required which is the biggest blocker at the moment.

> I am trying to work with PCBWay on how to get this done. 

See if you can find out what quantity they need for external parts.  I would imagine that it's probably a reasonably big number but if not, then you might be able to coordinate a small group buy.  E.G.: if they're willing to do a production run of only 10 boards using (some) customer supplied / 3rd party parts then you only need to find 10 boards worth of parts from which ever supplier has the parts that PCBWay does not have and ship them to PCBWay.

SpuriousGer, 2022-10-13:
I have received feedback from PCBWay. Attached is the Excel. I would require some help on this. Who could help? I think it looks promising overall.
[T-B2W570125A-5sets-view_base-bom(2022-10-13).xls](https://github.com/scottbez1/smartknob/files/9776060/T-B2W570125A-5sets-view_base-bom.2022-10-13.xls)

SpuriousGer, 2022-11-01:
I hit the next road bump (I guess to be expected as a complete novice), PCBWay is asking for the pick&place(centroid) file for the base PCB. Anyone any clue?

AminKAli, 2023-05-25:
I'd be interested in the group by if that was still happening. @SpuriousGer Let me know if I can help with anything

SpuriousGer, 2023-05-26:
I did order some PCBs from PCBWay and have all the parts, but I never got around to assembling it.
I'd still be interested in the group, I'd be happy to provide funding, but I would need someone to help with instructing PCB and assembling all the required parts.

percramer, 2023-08-08:
I am also interested and I work with pcbway often (also for cnc machining etc). Where did you get stuck with the board Assembly with them? The pick & place location file?

jantyikt, 2023-08-08:
After a few emails, everything was sorted. They used/found a bit of different parts here and there, the ones they had available (+ they couldn't get the heatsink and the tube, which is not a problem anyway). Their first letter was a bit misunderstood, I think, but then it all went about smoothly. The end product looks really good but I did not have time to try out.

sebastienvermeille, 2024-03-04:
Hi guys, did you manage to collect all the required files ? I would like to order also a presoldered pcb as SMD is too difficult for me to solder.

Thank you :)

varunkotian, 2024-09-26:
you can find it here. But these are unverified.

[bom.csv](https://smartknob-artifacts.s3.us-west-1.amazonaws.com/dev/assemblyFiles/electronics/view_base-jlc/bom.csv)
[pos.csv](https://smartknob-artifacts.s3.us-west-1.amazonaws.com/dev/assemblyFiles/electronics/view_base-jlc/pos.csv)