scottbez1/smartknob #24: Use FSR instead of Strain Gauge

Chaphasilor, 2022-03-19:
Hi there,

as you've mentioned multiple times in the readme, working with strain gauges isn't trivial, so it's easy to screw up when mounting them to the PCB. You can practice mounting them somewhere else beforehand of course, but that means ordering more units than you'd actually need.

I'm currently working on a project where I also need to detect pressure being applied to something, and I've found that Force-Sensitive Resistors (FSRs) are quite easy to work with and offer a straight-forward implementation when it comes to measuring force.  
[Here's an example of what I'm talking about](https://www.aliexpress.com/item/32960696900.html?gatewayAdapt=glo2deu&spm=a2g0o.productlist.0.0.52441640oMyG0o&algo_pvid=77bb19d9-2a18-4792-bac2-e933730d2269&algo_exp_id=77bb19d9-2a18-4792-bac2-e933730d2269-0&pdp_ext_f=%7B%22sku_id%22%3A%2266555936662%22%7D&pdp_pi=-1%3B1.98%3B-1%3B-1%40salePrice%3BEUR%3Bsearch-mainSearch)
They usually are a bit more expensive, but as I mentioned, working with them is probably easier than using strain gauges.

Maybe give them a try? :)

st157322, 2022-03-21:
Even easier would be this solution: https://github.com/IvDm/Z-probe-on-smd-resistors-2512

rtbcoop, 2022-04-08:
Or, replace the display with a touch screen.  Then you have no need for the strain gauge/FSR and you open the project up to a ton more features such as gesture controls.

omdathetkan, 2022-04-08:
Another thing to consider is inductive sensing (for example the TI LDC3113). This allows measuring micrometer level displacement (to a metal target).

PR #104 (2022-11-27): Use 2512 SMD resistors in place of strain gauges
As suggested in https://github.com/scottbez1/smartknob/issues/24#issuecomment-1073540944 and inspired by https://github.com/IvDm/Z-probe-on-smd-resistors-2512, this adds 4x 330 ohm 2512 SMD resistors to the back to act as strain gauges, rather than requiring discrete strain gauges to be glued and wired up.

Also applies a few changes I accidentally had pending locally:
- Custom footprints for the 2 3.3v LDO options to clarify the pinout options
[image]
- some BOM note updates

Fixes #24
Closes #33
commit 340eed93 (2022-11-27): Use 2512 SMD resistors in place of strain gauges (#104)

As suggested in https://github.com/scottbez1/smartknob/issues/24#issuecomment-1073540944 and inspired by https://github.com/IvDm/Z-probe-on-smd-resistors-2512, this adds 4x 330 ohm 2512 SMD resistors to the back to act as strain gauges, rather than requiring discrete strain gauges to be glued and wired up.

Also applies a few changes I accidentally had pending locally:
- Custom footprints for the 2 3.3v LDO options to clarify the pinout options
[image]
- some BOM note updates

Fixes #24
Closes #33
closed 2022-11-27 by scottbez1