scottbez1/smartknob #27: 32 mm BLDC Motor isn't for sale anymore :(

18Markus1984, 2022-03-26:
Hello, I saw your video yesterday for the first time and was blown away. I always wanted such a thing, but DIY and not like the Nest thing. I directly made a shopping list for all parts and had no problem to find all the parts besides the BLDC Motor. And also the price for all parts with shipping to Germany is around €85, pretty good. And if the shipping cost for the PCBs weren't that high, it would be even cheaper.

Likewise, I first used the link to find the motor isn't for sale anymore. So I thought maybe find it somewhere else on the internet, but... google pictures said No. I really tried to find the thing with all the product pictures the names of the specifications, but I didn't find it. 

The only motor of similar size I could spot was a bit bigger [one](https://de.aliexpress.com/item/1005003751799249.html?gatewayAdapt=glo2deu&spm=a2g0o.detail.0.0.6bf048a7rY6zH3&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.274681.0&scm_id=1007.13339.274681.0&scm-url=1007.13339.274681.0&pvid=df4ad597-8a5a-450c-8475-9387ba4b90fc&_t=gps-id:pcDetailBottomMoreThisSeller,scm-url:1007.13339.274681.0,pvid:df4ad597-8a5a-450c-8475-9387ba4b90fc,tpp_buckets:668%232846%238111%231996&pdp_ext_f=%257B%2522sku_id%2522%253A%252212000027041443598%2522%252C%2522sceneId%2522%253A%25223339%2522%257D&pdp_pi=-1%253B3.07%253B-1%253B-1%2540salePrice%253BEUR%253Brecommend-recommend)

I could use the [smaller](https://de.aliexpress.com/item/1005003240145615.html?gatewayAdapt=glo2deu&spm=a2g0o.order_list.0.0.21ef5c5fvbXmAc) motor, but you said on your other branch "moderate cogging - not ideal for completely smooth input". I ordered this one, but do you think it is relatively easy to remodel the design for the 34 mm Version? Or is there definitely not enough space, so I had to remodel the complete exterior covering? 
Maybe you found a better motor till now.

Happy to hear from you 👍

ioma8, 2022-03-27:
I am also interested where to get such motor.

PaulGoldschmidt, 2022-03-28:
Hey @18Markus1984, I'm also quite interested in building such a knob - additionally, I'm living in Germany (Heidelberg).
If you are interessted in a collaboration, please let me know.

Kind regards,
Paul

C44Supra, 2022-03-28:
@18Markus1984 There's a ticket for the motor stocking issue already: https://github.com/scottbez1/smartknob/issues/16
PCBs shouldn't be that expensive, you can get them made for a couple of dollars and shipped for a few more dollars at JLC. All depends on how patient you are.

18Markus1984, 2022-03-28:
> @18Markus1984 There's a ticket for the motor stocking issue already: #16 PCBs shouldn't be that expensive, you can get them made for a couple of dollars and shipped for a few more dollars at JLC. All depends on how patient you are.

Yes I always order at JLC but if you look at the recent [shipping prices](https://photos.app.goo.gl/VT93cuQ5iBZmSmxy6). You can see that they got a little bit higher. Lowest 16$

18Markus1984, 2022-03-28:
> Hey @18Markus1984, I'm also quite interested in building such a knob - additionally, I'm living in Germany (Heidelberg). If you are interessted in a collaboration, please let me know.
> 
> Kind regards, Paul

Ja wäre nice. Schreib mich einfach mal an.

C44Supra, 2022-03-28:
@18Markus1984 @PaulGoldschmidt I'm also in Germany. My PCBs were delivered earlier today. Most of the components I have on hand as well. Just waiting for the AliExpress parts. Problem is that with a different motor there might be some mounting issues.

18Markus1984, 2022-03-28:
> @18Markus1984 @PaulGoldschmidt I'm also in Germany. My PCBs were delivered earlier today. Most of the components I have on hand as well. Just waiting for the AliExpress parts. Problem is that with a different motor there might be some mounting issues.

So you have a spare one? That I don't have to order them?

C44Supra, 2022-03-28:
I have a spare set of PCBs yes. I should have spare components too. I can put the components on for you as well. Probably a good idea with the TMC6300. If you want to put the rest on yourself, I'll leave that up to you.

Horia73, 2022-03-28:
I am also interested in building this knob. If it s someone here that have multiple pcbs and spare parts and wants to sell them please let me know. I m from Romania.

zablotzky, 2022-03-28:
> I have a spare set of PCBs yes. I should have spare components too. I can put the components on for you as well. Probably a good idea with the TMC6300. If you want to put the rest on yourself, I'll leave that up to you.

Oh I would be happy to get one with the components as well! How can we get in touch to get this organised?

Horia73, 2022-03-28:
> Oh I would be happy to get one with the components as well! How can we get in touch to get this organised?

Me too :))

C44Supra, 2022-03-28:
I should have about 3 spares as I'm looking to keep 2 for myself. In hindsight, probably should've ordered a few more PCBs. Either way, don't want to step on @scottbez1's toes, if he's okay with me charging a little bit of money for assembly on top of PCB + parts + shipping, we can go ahead.

18Markus1984, 2022-03-28:
> I should have about 3 spares as I'm looking to keep 2 for myself. In hindsight, probably should've ordered a few more PCBs. Either way, don't want to step on @scottbez1's toes, if he's okay with me charging a little bit of money for assembly on top of PCB + parts + shipping, we can go ahead.
> 
> That said, I'm still waiting for the ESPs to come in. According to AliExpress shipping that should be this week. Once those are here I should have all the parts needed for PCBA.

Yes, that would be pretty nice. I will definitly pay you for this. My mail is markusschmidt1984@gmx.de

zablotzky, 2022-03-28:
From experience with other projects devs usually have nothing against it, as long as feedback is provided to keep track of all the issues. Drop me an email at jacek dot zablocki at outlook dot com to push things forward :)

Horia73, 2022-03-28:
> I should have about 3 spares as I'm looking to keep 2 for myself. In hindsight, probably should've ordered a few more PCBs. Either way, don't want to step on @scottbez1's toes, if he's okay with me [...]

I will definitely  pay you. My email is iugahoria@gmail.com 
If @scottbez1 it s okay with this of course.

C44Supra, 2022-03-28:
@18Markus1984 @zablotzky @Horia73 Sent you guys some emails.

That motor does indeed look to be the same one. Never heard of that specific site before though. I they have stock, it's likely limited though. I'm currently in touch with a manufacturer for possible replacement motors as per #16

C44Supra, 2022-03-28:
Looks to only ship within India as well, that could be an issue for anyone outside of India.

scottbez1, 2022-03-28:
> don't want to step on @scottbez1's toes, if he's okay with me charging a little bit of money for assembly on top of PCB + parts + shipping, we can go ahead.

Go for it - that's the beauty of open source!

18Markus1984, 2022-03-28:
> @18Markus1984 There's a ticket for the motor stocking issue already: #16 PCBs shouldn't be that expensive, you can get them made for a couple of dollars and shipped for a few more dollars at JLC. All depends on how patient you are.

18Markus1984, 2022-03-28:
> @18Markus1984 There's a ticket for the motor stocking issue already: #16 PCBs shouldn't be that expensive, you can get them made for a couple of dollars and shipped for a few more dollars at JLC. All depends on how patient you are.

Extra looked over all the tickets, but sadly I missed it  😅

closed 2022-03-28 by 18Markus1984

18Markus1984, 2022-05-08:
> @18Markus1984 @zablotzky @Horia73 Sent you guys some emails.
> 
> That motor does indeed look to be the same one. Never heard of that specific site before though. I they have stock, it's likely limited though. I'm currently in touch with a manufacturer for possible replacement motors as per #16

I accidentally deleted your mail and didn't find a better way to ask when I could pay you for the PCBs?