scottbez1/smartknob #29: can't find ESP32 when upload firmware to base board.

zinwalin, 2022-03-28:
Building in release mode
Retrieving maximum program size .pio\build\view\firmware.elf
Checking size .pio\build\view\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.4% (used 17648 bytes from 327680 bytes)
Flash: [===       ]  27.3% (used 357450 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM3
Uploading .pio\build\view\firmware.bin
esptool.py v3.1
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2
================================================================================================================================ [FAILED] Took 28.07 seconds ================================================================================================================================
Environment    Status    Duration
-------------  --------  ------------
view           FAILED    00:00:28.074
=========================================================================================================================== 1 failed, 0 succeeded in 00:00:28.074 ===========================================================================================================================The terminal process "C:\Users\zinwa\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'view'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

[image]

scottbez1, 2022-03-28:
It looks like Q2 is not populated, which is the dual transistor package that provides the reset and boot mode selection via USB serial DTR and RTS, so automatic downloads won't work without that.

[image]

You can use the solder pads for EN and BOOT on the bottom to manually put the chip into program mode - connect BOOT to ground, then start the upload, then briefly connect EN to ground while you see the Connecting..... message.

zinwalin, 2022-03-28:
> 

@scottbez1 Hi Scott, thanks for your reply, Q2 component is not shipped out yet, I used EN and BOOT to flash once successfully, it's quite tricky.   BTW, where did you find the picture you referred in your reply?

scottbez1, 2022-03-28:
Awesome, glad you were able to get it working! That photo is from the [schematic](https://smartknob-artifacts.s3.us-west-1.amazonaws.com/refs/tags/releases/view_base/v0.4/electronics/view_base-schematic.pdf) which can be found linked from the [releases](https://github.com/scottbez1/smartknob/releases) or on the main README (for the latest bleeding edge version)

closed 2022-03-28 by scottbez1