scottbez1/smartknob #30: Can you upload real photos for your soldered PCB boards?

zinwalin, 2022-03-29:
I purchased components from online shop, when soldering it's quite tricky to get the right direction for LEDs, need your PCB boards for reference. Appreciated for your efforts.

C44Supra, 2022-03-29:
There are links in the Readme to the InteractiveHTMLBOM for both the main PCB (https://smartknob-artifacts.s3.us-west-1.amazonaws.com/master/electronics/view_base-ibom.html) and the screen PCB (https://smartknob-artifacts.s3.us-west-1.amazonaws.com/master/electronics/view_screen-ibom.html). If you press the gear icon in the top right of the page and select the option "Highlight first pin", that will show you part polarization.

Most of the components speak for themselves, but for LEDs and diodes in general, pin one is cathode. Resistors and caps are non-polarized so no worries there.

scottbez1, 2022-03-29:
If you haven't already, I'd recommend opening the [view_base](https://github.com/scottbez1/smartknob/tree/59420c6da924c546f69fda3c9bdcbbe309c0524d/electronics/view_base) project in KiCad and using the built-in 3d viewer to look around. I took care to make sure as many components as possible had accurate 3d models associated with them so the render should be very accurate.

[image]

Alternatively, the main [3d mechanical cad file](https://a360.co/3BzkU0n) includes a semi-recent export of that PCB 3d model, so you can use Autodesk's web-based 3d viewer to look around at the PCB.

[image]

Let me know if you still have questions after checking out those tools, and I can snap some photos of specific areas if there's something that's not clear.

zinwalin, 2022-03-30:
> 

Thanks.  U1, U3, U4, U5 have polarized restriction, I made mistake for U5 at first soldering, and when powered, TMC6300 is so hot. After change the direction, it works.

closed 2022-03-30 by scottbez1