scottbez1/smartknob #32: Current draw

psyspy1, 2022-03-29:
Hi! Thanks for open sourcing this project. Looks amazing! I am looking into integrate this into my hobby IoT project. I am curious what the current consumption. My IoT project takes in power through usb 2.0, so in total 500 mA.

scottbez1, 2022-03-29:
It will depend a lot on the specific motor you use. I don't have a current draw measurement handy, but based on the coil resistance I'd expect somewhere between 700-1000mA max current draw with this one (though this specific motor sadly does not seem to be available any longer... #16)

Keeping under 500mA between the motor and any wifi/BT radio usage is likely to be tricky (though you can tweak the max voltage in the SimpleFOC configuration code to proportionally reduce the max current, at the expense of less torque).

If it's a possibility for you, I'd suggest looking at using a type C USB 2.0 connector (like the one used in the project) with 5.1k resistors to ground on the CC lines, which should allow for 5V up to 3A with a proper type C power supply. So you get the relative simplicity of USB 2.0, but higher current than a mini/micro connector, without having to deal with all the extra pins of a full USB 3 type C port or PD negotiation. https://www.scorpia.co.uk/2016/03/17/using-usb-type-c-on-hobyist-projects/

[image]

zinwalin, 2022-03-30:
> 

> It will depend a lot on the specific motor you use. I don't have a current draw measurement handy, but based on the coil resistance I'd expect somewhere between 700-1000mA max current draw with this [...]

Hi Scott, is the current big enough to connect with PC using USB?

scottbez1, 2022-03-30:
I have done most of my testing connected to a PC. As long as you use a Type-C to Type-C cable (do NOT use an A to C cable or other adapters!) and your computer is capable of supplying 1A+ at 5v you should be fine.

scottbez1, 2022-04-05:
Closing for now. Let me know if you have other questions.

closed 2022-04-05 by scottbez1