scottbez1/smartknob #36: How to detect wheter motor and driver is working properly?

zinwalin, 2022-03-30:
I used TMC6300 and 3215 motor, looks like there's no torque when screen PCB is powered.

zinwalin, 2022-03-30:
Looks like the motor is not working at all.

---------------------

--- Miniterm on COM5  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
MotorTask::run()
MOT: Monitor enabled!
MOT: Init
MOT: Enable driver.
millilux: 2340.00
MOT: Align sensor.
MOT: Skip dir calib.
MOT: Skip offset calib.
millilux: 2318.40
MOT: No current sense.
MOT: Ready.
Press Y to run calibration
millilux: 2433.60
millilux: 2534.40
millilux: 2599.20
2.93
Got new config
millilux: 2548.80
millilux: 2577.60
millilux: 2577.60
millilux: 2599.20
millilux: 2592.00

scottbez1, 2022-03-30:
Have you run calibration yet? Press Y in the serial monitor when you see the "Press Y to run calibration" message at startup (you have to be somewhat quick). This will run through a few open-loop movements of the motor (that don't require any other sensors or components to work correctly, unlike normal operation). 

If the motor doesn't work during that, then you'll need to debug further - probably best to oscillocope probe the UVW test points on the bottom for that, to see if there is any voltage coming out of the TMC6300. You could also probe R7-R12 to verify that the correct signals for UL/UH/VL/VH/WL/WH are being output by the ESP32.

But also, in #34 you mentioned that the TMC6300 got very hot due to being soldered in the wrong orientation - did you replace that chip with a fresh one, or just rotate it?

zinwalin, 2022-03-31:
> Have you run calibration yet? Press Y in the serial monitor when you see the "Press Y to run calibration" message at startup (you have to be somewhat quick). This will run through a few open-loop [...]

Hi Scott, when goes into calibration, the fireware crashed and base PCB reboot.  In terms of TMC6300, I only rotate chip to make it run without hot. 

Here's the crash log when calibration.

--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
millilux: 3132.00
MT6701 getSensorAngle...
0.15
millilux: 3124.80
Press Y to run calibration
millilux: 3153.60
millilux: 3225.60
millilux: 3304.80
millilux: 3355.20
millilux: 3384.00
millilux: 3484.80

Sensor measures positive for positive motor rotation: NO, Direction=CCW
millilux: 3564.00
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (Motor)
Core 1 register dump:
PC      : 0x400db5d1  PS      : 0x00060c36  A0      : 0x800e0d35  A1      : 0x3ffb8280  
A2      : 0x00000000  A3      : 0x00000011  A4      : 0x00000001  A5      : 0x3f4003b8
A6      : 0x3ffb8660  A7      : 0x00000008  A8      : 0x3f4003a7  A9      : 0x3ffb8474  
A10     : 0x3ffb8290  A11     : 0x3ff9c510  A12     : 0x000000ff  A13     : 0x0000ff00
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x400db5d1:0x3ffb8280 0x400e0d32:0x3ffb8590 0x400e0d6a:0x3ffb8620 0x400d6672:0x3ffb8660 0x400d2808:0x3ffb8700 0x400d22a9:0x3ffb8850 0x40088211:0x3ffb8870
  #0  0x400db5d1:0x3ffb8280 in _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:929
  #1  0x400e0d32:0x3ffb8590 in _vsnprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:72
  #2  0x400e0d6a:0x3ffb8620 in vsnprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:41
  #3  0x400d6672:0x3ffb8660 in Print::printf(char const*, ...) at C:\Users\zinwa\.platformio\packages\framework-arduinoespressif32\cores\esp32/Print.cpp:254
  #4  0x400d2808:0x3ffb8700 in MotorTask::run() at src/motor_task.cpp:163
  #5  0x400d22a9:0x3ffb8850 in Task<MotorTask>::taskFunction(void*) at src/task.h:46
  #6  0x40088211:0x3ffb8870 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jul 29 2019 12:21:46

zinwalin, 2022-03-31:
BTW, another question for TMC6300, how to check whether it is detected or is running well?

zinwalin, 2022-03-31:
> 
Hi Scott, good news for the issue, after checkc R7-R12, and U/V/W for motor, ESP32 output is normal, U/V/W is zero, so I replaced a new TMC6300, and it is powered, the motor moves.   Brilliant, thanks for your tips.

closed 2022-03-31 by scottbez1