scottbez1/smartknob #37: crash when run calibration with typing Y

zinwalin, 2022-03-31:
--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Sprite created!
millilux: 26236.80
millilux: 26949.60
Press Y to run calibration
Ymillilux: 26942.40
millilux: 27057.60
millilux: 26697.60
millilux: 27021.60

Sensor measures positive for positive motor rotation: YES, Direction=CW
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (Motor)
Core 1 register dump:
PC      : 0x400dfaf1  PS      : 0x00060c36  A0      : 0x800e5255  A1      : 0x3ffb8280
A2      : 0x00000000  A3      : 0x00000011  A4      : 0x00000001  A5      : 0x3f404396  
A6      : 0x3ffb8660  A7      : 0x00000008  A8      : 0x3f404385  A9      : 0x3ffb8474
A10     : 0x3ffb8290  A11     : 0x3ff9c510  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400dfaf1:0x3ffb8280 0x400e5252:0x3ffb8590 0x400e528a:0x3ffb8620 0x400da4ca:0x3ffb8660 0x400d3280:0x3ffb8700 0x400d2ced:0x3ffb8850 0x400883f2:0x3ffb8870
  #0  0x400dfaf1:0x3ffb8280 in _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:929
  #1  0x400e5252:0x3ffb8590 in _vsnprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:72
  #2  0x400e528a:0x3ffb8620 in vsnprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib

scottbez1, 2022-04-07:
It looks like this is probably the same issue as in #50 - can you try the fix suggested here: https://github.com/scottbez1/smartknob/issues/50#issue-1195172430

scottbez1, 2022-04-07:
Closing since I'm pretty sure this was the same issue as #50 and should be fixed in e35319a. Please let me know if increasing the stack size didn't fix it though.

closed 2022-04-07 by scottbez1

zinwalin, 2022-04-10:
@scottbez1 the commit https://github.com/scottbez1/smartknob/commit/e35319af6d4223d03087177da39b63936b2836fb didn't fix the issue. 

----

Sensor measures positive for positive motor rotation: YES, Direction=CW
millilux: 78379.20
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (Motor) 
Core 1 register dump:
PC      : 0x400dfb19  PS      : 0x00060c36  A0      : 0x800e527d  A1      : 0x3ffb8280
A2      : 0x00000000  A3      : 0x00000011  A4      : 0x00000001  A5      : 0x3f404463  
A6      : 0x3ffb8660  A7      : 0x00000008  A8      : 0x3f404452  A9      : 0x3ffb8474
A10     : 0x3ffb8290  A11     : 0x3ff9c510  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x400dfb19:0x3ffb8280 0x400e527a:0x3ffb8590 0x400e52b2:0x3ffb8620 0x400da4f2:0x3ffb8660 0x400d32cc:0x3ffb8700 0x400d2d01:0x3ffb8850 0x400883f2:0x3ffb8870
  #0  0x400dfb19:0x3ffb8280 in _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:929
  #1  0x400e527a:0x3ffb8590 in _vsnprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:72
  #2  0x400e52b2:0x3ffb8620 in vsnprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:41
  #3  0x400da4f2:0x3ffb8660 in Print::printf(char const*, ...) at C:\Users\zinwa\.platformio\packages\framework-arduinoespressif32\cores\esp32/Print.cpp:254
  #4  0x400d32cc:0x3ffb8700 in MotorTask::run() at src/motor_task.cpp:159
  #5  0x400d2d01:0x3ffb8850 in Task<MotorTask>::taskFunction(void*) at src/task.h:46
  #6  0x400883f2:0x3ffb8870 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jul 29 2019 12:21:46