scottbez1/smartknob #38: Idea: The BLDC motor could be used to implement inertial scrolling

RichardBronosky, 2022-03-31:
Imagine a scroll wheel that had the inertial scrolling commonly seen on multi-touch trackpads. But because the motion could be physically represented by the driven motor, the software could adapt to the user. The user could have dents until they fling it. The fling could have friction that lessens if the user give additional flings upon slowing.

Note: This is constitutes *Prior Art*. The feature described is currently in development and offered freely to the public domain and may not be used in patents.

leofidus, 2022-03-31:
This sounds a lot like what the Logitech MX series uses as a scroll wheel: at low speeds it has the usual detent-like behavior, fling it hard enough and it becomes free flowing with inertia; with a button to switch behavior manually. Although I think they use a purely mechanical solution to achieve this.

It's certainly a very nice feature that feels very satisfying.

RichardBronosky, 2022-03-31:
> This sounds a lot like what the Logitech MX series uses as a scroll wheel

I have used that mouse and you are correct, it is an extremely complex mechanical implementation. And it's mechanical properties limit it in that the spacing and depth of the detents are on software configurable nor is the mass as drag of the wheel. 

Related note:
I've been working on incorporating algorithms from mobile/gaming haptic libraries to produce different texture simulations. As is, values can be adjusted very small to change the "detents" to feel like "grit" or "scratch" or stretched and smoothed to feel like "waves". But I'd like to get to a point where the value (say 0-255) could be sensed in real time by the feel of the wheel while (and shortly after) turning. The LCD is a sexy feature, but for productivity, you shouldn't have to take your eyes off the screen to know what the LCD is telling you. 🤔💡

drivepilot, 2022-04-06:
I was intrigued to learn about the Logitech MX mouse and found this very interesting teardown which shows that in version 3 of the device it is actually magnetically enabled. https://blog.bolt.io/logitech-mx-master-3-vs-2s/

iggut, 2022-08-15:
Microsoft Arc Mouse uses some soft of motor to simulate rotational inertia is also a cool example of what can be done.