scottbez1/smartknob #45: [Imporvement] Hard to fix the LCD screen in the knob

zinwalin, 2022-04-05:
See the pciture depicted below, the screen moved when I assembled, there's not slot to fix the screen. 

[image]

scottbez1, 2022-04-07:
Hmmm, which part came loose - was it the joint between the MountBase and ScreenPlatform 3d printed parts, or was it the joint between the ScreenPlatform and the screen PCB?

In general there should not be any interference as long as the screen PCB assembly was assembled within tolerance (i.e. using 0.6mm VHB tape rather than the more common 1.1mm variant, making sure the LCD is fully attached to the PCB with no gaps from the wires, etc). Then as long as you are careful to align the spokes of the RotorSpacer to avoid the ribbon cable "chin" on the LCD when assembling, everything should fit together without issue.

If you used mechanical parts prior to v185, the gap between the LCD and the inside of the Knob was slightly too small, causing potential rubbing when the knob is rotated, but this would not have been enough to dislodge the LCD assembly. An additional 0.5mm gap between LCD and knob was added in the v185 release to account for that potential issue.

willumpie82, 2022-04-19:
dit you consider using a piece of brass tube and just solder that to upper and lower boards (perferable with dualsided exposed copper on the boards for rigidity)?

willumpie82, 2022-05-12:
simple proof of concept with a 5mm brass tube, I'll try to create a version of the boards to have a coper area to solder the shaft on, a bit of grease can be added to smooth the rotation, there is about .15mm clearance between the shaft and the motor hollow shaft radius
[image]
[image]