scottbez1/smartknob #48: How to check whether HX711/Strain gauges has problem?

zinwalin, 2022-04-06:
It's not working on my side, before fixed strain gauges on PCB, I just want to validate strain gauges is originally working with HX711. 

[image]

scottbez1, 2022-04-07:
The strain gauge thresholds will need to be manually calibrated based on your physical assembly (every build will be different).

The code outputs a `HX711 reading` value to the serial console once per second by default. You can check that this changes when you touch a strain gauge.

Then once the strain gauges are permanently installed, you can get a reading when nothing is pressed and set this as the "lower" value, and then press down on the knob and get a reading there and set that as the "upper" value here: https://github.com/scottbez1/smartknob/blob/5c2a10428a7c4736a4aa2c93c831fbd3f2271b72/firmware/src/interface_task.cpp#L206-L207

scottbez1, 2022-04-07:
Also, by the way, for testing the knob without the strain gauges, you can press [space] in the serial monitor to switch to the next mode in lieu of activating the strain gauges.

zinwalin, 2022-04-07:
> Also, by the way, for testing the knob without the strain gauges, you can press [space] in the serial monitor to switch to the next mode in lieu of activating the strain gauges.

Yes. I get used to press space to change mode.

zinwalin, 2022-04-07:
> The strain gauge thresholds will need to be manually calibrated based on your physical assembly (every build will be different).
>
> The code outputs a `HX711 reading` value to the serial console once [...]

Hi Scott, your reply saved lots of time, there's no console output for HX711 reading by default, cause the lower and upper values are too high on my side, so the logic for HX711 is not run.  After I lower the values, it works.  Appreciated for your reminders.

scottbez1, 2022-04-07:
Glad to have helped!

closed 2022-04-07 by scottbez1