scottbez1/smartknob #50: Debug exception reason: Stack canary watchpoint triggered (Motor)

shirogeek, 2022-04-06:
Hi,

First thanks a lot for this amazing piece of DIY. Great to get a hands on feeling of haptic feedback. 
I am having a little trouble flashing the firmware and getting it to work. First, on the master branch only the view env compiles (which I believe is normal looking at the dependencies). After flashing the view build I get stuck in the following bootloop :

`ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Sprite created!
millilux: 1764.00
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (Motor) 
Core 1 register dump:
PC      : 0x400dfa22  PS      : 0x00060836  A0      : 0x800e51e5  A1      : 0x3ffb8220  
A2      : 0x00000000  A3      : 0x3ffb8414  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ffb8600  A7      : 0x00000008  A8      : 0x3ffb8530  A9      : 0x3ffb8210  
A10     : 0x3ffb8230  A11     : 0x3ff9c510  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000020  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400dfa22:0x3ffb8220 0x400e51e2:0x3ffb8530 0x400e521a:0x3ffb85c0 0x400da45a:0x3ffb8600 0x400d3cfb:0x3ffb86a0 0x400d5218:0x3ffb86d0 0x400d3152:0x3ffb8700 0x400d2cdd:0x3ffb8850 0x400883f2:0x3ffb8870

Rebooting...
`

Should I assume bad soldering on the TMC6300 ? Is there something else I am missing ?

Thanks a lot

scottbez1, 2022-04-07:
Yeah `view` is the correct env, I should clean up the others.

Given `Stack canary watchpoint triggered (Motor)` message it looks like the motor task stack is not large enough (interesting, since it works fine for me).

Try increasing the stack size defined in the MotorTask constructor [near the top of motor_task.cpp](https://github.com/scottbez1/smartknob/blob/5c2a10428a7c4736a4aa2c93c831fbd3f2271b72/firmware/src/motor_task.cpp#L19), from 1200 to maybe 2048 which should be way more than enough?

shirogeek, 2022-04-07:
Thanks a lot.
I have moved past that bug. I will continue the tests and playing with it tonight. Will let you guys know how it goes. Again, thanks for the amazing work !

scottbez1, 2022-04-07:
Awesome, thanks for following up! I will go ahead and increase that default stack size. I think I may have shrunk it when trying to get wifi working, but since that's looking like it will require a bigger change (to lvgl to free up the need for a full framebuffer in memory), might as well undo that optimization for now since it's causing issues.

closed 2022-04-07 by scottbez1