scottbez1/smartknob #52: Move hardcoded configurations to a file on filesystem

scottbez1, 2022-04-07:
There are a number of hardcoded configuration values currently:
- electrical zero angle (motor + sensor alignment)
- motor pole count
- motor orientation
- HX711 upper and lower thresholds
- VEML7700 lux mapping to LED/backlight brightness

Might be nice to move these to a file (json?) on spiffs/littlefs, and ideally eventually add a calibration menu that can write/update that file automatically.

mcmadhatter, 2022-04-07:
JSON on an spiffs is a great idea.
If wifi is working then the calibration json file can also be written via a webpage running on the esp32, with a nice interface for setting the values. Here is an example of where i’ve done something similar in the past https://github.com/mcmadhatter/LocoNet2/tree/development/examples/esp-locoio-web

mcmadhatter, 2022-05-08:
I have started work on this on my fork , currently just the knob configurations are supported. Next steps is to add support for information that can be gathered via auto calibration.

https://github.com/mcmadhatter/smartknob/tree/JSON

septatrix, 2022-06-24:
Adding a JSON library would add significant code size. If setting the values is exposed by the firmware (custom menu, webpage etc.) it is probably more convenient to just dump a C struct into a file. Also the ESP32 provides some pseudo-EEPROM which is ideal for calibration values etc without the overhead of a whole filesystem given that I expect no one to manually flash those setting on a SD card or similar.

leofidus, 2022-06-24:
If an actual file is desired an INI file might be much better than JSON: easier to parse in code, easier to read for humans.

I tend to agree though that a file might be overkill, dumping a C struct to non-volatile storage might be enough