scottbez1/smartknob #56: The board restarts indefinitely

736676938, 2022-04-07:
LOG:

> Executing task in folder test: C:\Users\wlj\.platformio\penv\Scripts\platformio.exe run --target upload --target monitor <

Processing view (board: esp32doit-devkit-v1; platform: espressif32@3.4; framework: arduino)
---------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (3.4.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Simple FOC> 2.2.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <TLV493D-Magnetic-Sensor> 1.0.3
|   |-- <Wire> 1.0.1
|-- <AceButton> 1.9.1
|-- <TFT_eSPI> 2.4.25
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <FastLED> 3.5.0
|   |-- <SPI> 1.0
|-- <HX711> 0.7.5
|-- <Adafruit VEML7700 Library> 1.1.1
|   |-- <Adafruit BusIO> 1.11.3
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
Building in release mode
Retrieving maximum program size .pio\build\view\firmware.elf
Checking size .pio\build\view\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.4% (used 17648 bytes from 327680 bytes)
Flash: [===       ]  27.3% (used 357434 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM6
Uploading .pio\build\view\firmware.bin
esptool.py v3.1
Serial port COM6
Connecting......
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:75:25:a8:34:50
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
[... 248 lines cut ...]
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400f2040: ffff5755 ffffffff ffffffff
Core 1 register dump:
PC      : 0x400f2044  PS      : 0x00060230  A0      : 0x0000000d  A1      : 0x3ffb8660
A2      : 0x00000021  A3      : 0x00000000  A4      : 0x0000001a  A5      : 0x3ff49028  
A6      : 0x0000001a  A7      : 0x00000001  A8      : 0x00002a00  A9      : 0x00000a00  
A10     : 0x0000000a  A11     : 0x3ff000c4  A12     : 0x06ff1f78  A13     : 0x3ffb1f60  
A14     : 0x3ffb8058  A15     : 0x00000001  SAR     : 0x0000000e  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400f2044:0x3ffb8660 0x0000000a:0x3ffb8690 0x400d5f67:0x3ffb86d0 0x400d3118:0x3ffb8700 0x400d2ced:0x3ffb8850 0x400883f2:0x3ffb8870
  #0  0x400f2044:0x3ffb8660 in mcpwm_gpio_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/mcpwm.c:699
  #1  0x0000000a:0x3ffb8690 in ?? ??:0
  #2  0x400d5f67:0x3ffb86d0 in BLDCDriver6PWM::init() at .pio/libdeps/view/Simple FOC@2.2.0/src/drivers/BLDCDriver6PWM.cpp:61
  #3  0x400d3118:0x3ffb8700 in MotorTask::run() at src/motor_task.cpp:61
  #4  0x400d2ced:0x3ffb8850 in Task<MotorTask>::taskFunction(void*) at src/task.h:46
  #5  0x400883f2:0x3ffb8870 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

scottbez1, 2022-04-07:
It looks like you do not have the correct ESP32 chip. I see this in your log: `Chip is ESP32-PICO-D4 (revision 1)`, but it should say `Chip is ESP32-PICO-V3-02`

Are you sure you got the "Plus" version of the T-Micro32 as linked in the BOM? It should have a red chip antenna on the upper right corner, not a white chip antenna in the upper left.

Here's a comparison image:
[image]

I'm not certain this is the issue, but the pinout is different between the modules, and some of the GPIO pins used are only available in the PICO-V3-02 but reserved for internal flash use on the PICO-D4.

736676938, 2022-04-08:
> 感谢您提供完整的输出，这很有帮助！您似乎没有 ESP32 芯片。我在您的日志中看到了这个：`Chip is ESP32-PICO-D4 (revision 1)`，但它应该正确地说`Chip is ESP32-PICO-V3-02`
> 
> 您确定您有BOM 中的 T-Micro32 链接吗？
> 
> 这是一个比较图像： [image]
> 
> 我可以使用这个通用输入输出接口，但是，但它们之间的问题不是分配给 PICO 的，但 PICO-D3 内部使用但不能在内部使用。

Thank you for your timely reply. I will replace the ESP32 chip later

closed 2022-04-08 by scottbez1

WangChun420, 2022-04-11:
my chip (ESP32-PICO-V3-02) is right, but my dev also restarts indefinitely,dont know why

LOG:
Processing view (board: esp32doit-devkit-v1; platform: espressif32@3.4; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (3.4.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Simple FOC> 2.2.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <TLV493D-Magnetic-Sensor> 1.0.3  
|   |-- <Wire> 1.0.1
|-- <AceButton> 1.9.1
|-- <TFT_eSPI> 2.4.25
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <FastLED> 3.5.0
|   |-- <SPI> 1.0
|-- <HX711> 0.7.5
|-- <Adafruit VEML7700 Library> 1.1.1
|   |-- <Adafruit BusIO> 1.11.3
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
Building in release mode
Retrieving maximum program size .pio\build\view\firmware.elf
Checking size .pio\build\view\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.4% (used 17648 bytes from 327680 bytes)
Flash: [===       ]  27.3% (used 357450 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM6
Uploading .pio\build\view\firmware.bin
esptool.py v3.1
Serial port COM6
Connecting........_____....._____..
Chip is ESP32-PICO-V3-02 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, Embedded PSRAM, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:0b:fb:6b:de:4c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00067fff...
Flash params set to 0x0230
Compressed 17104 bytes to 11191...
Writing at 0x00001000... (100 %)
Wrote 17104 bytes (11191 compressed) at 0x00001000 in 0.6 seconds (effective 221.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 431.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 728.2 kbit/s)...
Hash of data verified.
Compressed 357552 bytes to 190499...
Writing at 0x00010000... (8 %)
Writing at 0x00018f7f... (16 %)
Writing at 0x0002ae4c... (25 %)
Writing at 0x00031fa9... (33 %)
Writing at 0x000375fc... (41 %)
Writing at 0x0003ce1b... (50 %)
Writing at 0x000433f6... (58 %)
Writing at 0x0004bfc6... (66 %)
Writing at 0x0005192b... (75 %)
Writing at 0x000579a0... (83 %)
Writing at 0x0005d7f9... (91 %)
Writing at 0x000635eb... (100 %)
Wrote 357552 bytes (190499 compressed) at 0x00010000 in 4.6 seconds (effective 625.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
=============================================================================== [SUCCESS] Took 22.38 seconds ===============================================================================

Environment    Status    Duration
-------------  --------  ------------
view           SUCCESS   00:00:22.381
================================================================================ 1 succeeded in 00:00:22.381 ================================================================================

WangChun420, 2022-04-11:
this is my reboot log

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
ALS sensor not found!
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400f844e  PS      : 0x00060d30  A0      : 0x800d9f79  A1      : 0x3ffb2ef0  
A2      : 0x00000000  A3      : 0x00000011  A4      : 0x0000000f  A5      : 0x3ffbd2c0  
A6      : 0x00000000  A7      : 0x3ffb2080  A8      : 0x00000000  A9      : 0x3ffb2ed0  
A10     : 0x00000000  A11     : 0x3ffb827c  A12     : 0x00000000  A13     : 0x00000020  
A14     : 0x00000020  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000e  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400f844e:0x3ffb2ef0 0x400d9f76:0x3ffb2f10 0x400d2a82:0x3ffb2f30 0x400d2cf9:0x3ffb2f90 0x400883f2:0x3ffb2fb0

varunkotian, 2024-11-19:
I also have a similar issue>
My repeating logs look something like this.

Rebooting...
[W][esp32-hal-i2c.c:1419] i2cCheckLineState(): invalid state sda(15)=0, scl(8)=0
[E][esp32-hal-i2c.c:1434] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=0, scl=0
[W][esp32-hal-i2c.c:1419] i2cCheckLineState(): invalid state sda(15)=0, scl(8)=0
[E][esp32-hal-i2c.c:1434] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=0, scl=0
[E][vfs_api.cpp:64] open(): /ffat/config.pb does not exist
SmartKnob starting!

Serial mode: plaintext
Press 'C' at any time to calibrate motor/sensor.
Press 'S' at any time to calibrate Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401048c2  PS      : 0x00060130  A0      : 0x800dcd4d  A1      : 0x3ffb96d0
A2      : 0x00000000  A3      : 0x00000011  A4      : 0x00060120  A5      : 0x3ffb9798
A6      : 0x3ffb2cd8  A7      : 0x00000000  A8      : 0x00060023  A9      : 0x3ffb8058
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x8008935c  A13     : 0x3ffb9c20
A14     : 0x00000000  A15     : 0x3ffb8b60  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000e  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x401048c2:0x3ffb96d0 0x400dcd4a:0x3ffb96f0 0x400d3b0a:0x3ffb9710 0x400d40ce:0x3ffb9750 0x400d42c9:0x3ffb97f0 0x4008914a:0x3ffb9810

Rebooting...

I don't know if this is due to somethings not soldered properly.
This is my first time doing such a thing.