scottbez1/smartknob #58: Idea: Use a capacitive touch screen display

rtbcoop, 2022-04-08:
If you use a touch screen display, you would no longer need the strain gauge.  You would also open up a lot of new capabilities such as gesture controls.

scottbez1, 2022-04-08:
Yeah, that would definitely be cool! I considered it and intentionally omitted it from this first iteration for 2 reasons:
- There are some really nice looking high dpi AMOLED displays based on the RM69330 driver IC from smartwatches that can be found for < US$20 with built in capacitive digitizers, but they have 4x the total pixel count (which may make them tricky to drive with an ESP32's limited memory) and there isn't yet open-source driver support in e.g. TFT_eSPI. Some discussion can be found [here](https://www.eevblog.com/forum/microcontrollers/driving-a-round-amoled-display/msg3757085/?PHPSESSID=lc6r0je7aip0qfv27a7bsfo9b0#msg3757085). I'm also not sure how readily available these are or if there's just excess stock that will disappear quickly.
- mechanically things get more complex, since the screen now needs to sit *above* the knob/rotor, rather than the current approach where there's a watch crystal directly attached to the knob/rotor that rotates with it. This means there needs to be an external gap between the display and the rest of the knob/rotor, which I think would be tricky to make look good especially with 3d-printing and manual assembly tolerances

rtbcoop, 2022-04-08:
I am sure there are others that are better suited to solve the electrical issue, but I can definitely solve the mechanical issue.  Instead of using a hollow shaft motor, I'd mount the knob to two captured bearings and then drive the knob with gears instead of directly off the motor.  This would:
1. Give you much better resolution (more steps per rotation of the knob thanks to gearing)
2. Give you more motor options since you aren't limited to hollow shaft motors
3. Provide more internal room for supports and wire runs.

These should all be commercial off the shelf parts that wouldn't add too much complexity or cost.  I'm glad to do the mechanical design legwork if this is a road you'd be interested in going down.  I'll post this up in Issue 16 as well to see if anyone has feedback.

DavidClawson, 2022-04-08:
I would wonder about the sensory feel of the drive gears. If the ratios were too high, it might be hard or strange to drive them by turning (vs a brushless which offers essentially zero resistance when not energized). You might also have issues with backlash, but maybe not too big of a deal.

hofstee, 2022-04-08:
> 1. Give you much better resolution (more steps per rotation of the knob thanks to gearing)

Magnetic encoders will already get you <0.1 degree usable resolution so I'm not sure having more than that will be meaningfully useful.

rtbcoop, 2022-04-09:
> I would wonder about the sensory feel of the drive gears. If the ratios were too high, it might be hard or strange to drive them by turning (vs a brushless which offers essentially zero resistance when not energized). You might also have issues with backlash, but maybe not too big of a deal.

Valid concern.  I don't think it would be terrible if you kept the gear ratio low.  That would take some testing.

> Magnetic encoders will already get you <0.1 degree usable resolution so I'm not sure having more than that will be meaningfully useful.

True, I don't think the added resolution would have any benefit on the sensing side, but I think it would make a difference on the feedback side.  You could have much finer detents and a smoother knob resistance.

rtbcoop, 2022-04-09:
This is what I had in mind.  In the real design, I would mount the gears to the bottom of the knob; this design was just for a visual. 

[image]

[image]

psyspy1, 2022-04-10:
@scottbez1 About your first point of the RM69330 touch screen display having higher pixel count and needing more memory, wouldn't [this](https://www.buydisplay.com/240x240-round-ips-tft-lcd-display-1-28-inch-capactive-touch-circle-screen) display work? It uses the same driver GC9A01A and seems to have the same dimensions and pixel count as the current one.

EUA, 2023-01-27:
Capacitive/Resistive touchscreen will also make **AC/DC converter and flex gauge sensors obsolete**. Make project easier and probably cheaper to build...