scottbez1/smartknob #59: Idea: Make version for retrofit in regular wall dimmer

omdathetkan, 2022-04-08:
I think it would be really great if this project would fit inside a regular wall mounted dimmer enclosure:
[image]

The shafts are standardized, and you would be able to combine it with the switch line of choice. This allows for the knob to blend in nicely with the rest house.

For me this would make the perfect KNX/Zigbee/ESPHome dimmer. If the motor sourcing issues are resolved I would be interested in doing some development on this.

scottbez1, 2022-04-08:
Do you have a reference for standard electrical box dimensions in your country? I assume these will vary a bit by region, but hopefully they are similar enough that there could just be a few alternate 3d printed front and back plate designs for different types but keep the core electronics/mechanics unchanged?

In the US a double-gang (square) switch box is 4"x4" (101.6mm) and the faceplate is generally 4.5"x4.5" (114.3mm), so the existing front plate (85mm) is not currently large enough.

The other things to figure out are power/connectivity in lieu of the USB-C connection on side, and maybe an alternative option for the ambient light sensor (currently down-firing to measure reflected light from the mounting surface - see photo)
[image]

Others have mentioned PoE for use in installations like this (#14) - I'm still not sure if it makes sense to add that to the base PCB (and there's the gpio constraint mentioned in that issue too), but maybe we could put a connector on the back with 5v, 3v, and UART pins and then develop a PoE "shield" that can be added (and would communicate via UART with the main MCU)?

Just throwing some ideas out there.

omdathetkan, 2022-04-09:
KNX is a single twisted pair that has both data and power. It’s a commonly used standard in residential and commercial building automation. 

For me the key was to use the smart knob with traditional/off the shelf face plates like these:
KNX is a single twisted pair that has both data and power. It’s a commonly used standard in residential and commercial building automation. 

For me the key was to use the smart knob with traditional/off the shelf face plates like these:
[image]

Making a custom pcb is a lot cheaper than custom plastic parts that look and feel good. 
You are right that different regions might require different designs.

I’ll see if I can make some CAD sketches sometime the coming days.

vlad-the-compiler, 2022-05-05:
To accommodate for most of the electrical boxes around the world, maybe an idea would be to get the smallest possible assembly for the knob mechanism and adapt it on a case-by-case basis for every popular standard, via a 3D printed bracket. Apart from material strength, aesthetics wouldn't matter for the bracket since it'd go inside the wall, and when it comes to the outward-facing faceplate, I reckon that could cheaply be made from CNC'd pieces of flat plastic.