scottbez1/smartknob #60: [improvement] hall sensor to scense press

gammeflamme, 2022-04-09:
This may be a stupid assumtion but would it be possible mount the hall sensor separatly to the motor and then when the pcb deflects from being pressed the hall sensor detects the increase in magnetic flux and triggers the klick function. Like the MT6701 datasheet discribes as "push button".

scottbez1, 2022-04-09:
Yeah, I saw that but wasn't able to get the push functionality on the
MT6701 working when I tested with a handheld magnet. I'm not sure why.

dorucazan, 2022-05-05:
Great work Scott! Working on a thermostat similar with Nest myself and I think that MT6701 could provide more accurate rotation data compared with TMAG5111; and probably the magnet could be cheaper than NMT-10-STIC magnetic tape from supermagnete.de. Thank you.
Regarding the push button functionality, there are 3 registers mentioned in the data sheet: PUSH_THRD, PUSH_DIFF_DLY  and PUSH_TIME_OUT but was not able to find their addresses and bit masks. Do you have more details about the MT6701 registers mapping? Many thanks!

dorucazan, 2022-05-06:
Just received an answer from MagnTek but it is not exactly the answer I was hoping for :confused: 

[image]