scottbez1/smartknob #61: Do I need to modify the code to make the new motor match this project?

stanlee1024, 2022-04-12:
Hi 

I had bought a new 2804 motor for this project, and also modify the 3D model to match the new motor. But when I put all the things together, it seems there is a little problem. I can't change the number when I rotate the knob, and the dot on the screen is shaking.
Do I need to modify the code to make the motor match this project?

stanlee-1024, 2022-04-12:
here is a picture of the new motor:

[image]

Here is a video shows what happened:

https://user-images.githubusercontent.com/102572250/162939774-5f0d2a43-4044-482e-9231-51f891a661e5.mp4

Paulsvcrch, 2022-04-12:
Do you have a magnet on the motor in order to give position feedback?
It seems to no count properly

scottbez1, 2022-04-13:
Yeah, if you haven't added a diametric magnet to that motor it's not going to be able to determine the rotor angle...

If you did add a magnet, then something is not working right with the sensor or the magnet - in Unbounded mode the motor isn't involved at all so it should just work regardless of the motor's specifics.

But generally speaking, you will need to run the calibration procedure for any motor (even if it was the same model as the original) - there's a serial prompt to run calibration at startup, and it will automatically determine the number of poles and the relative alignment of the mechanical and electrical angles through a series of movements. Right now the calibration process just outputs those constants to the serial monitor when it's done; you'll need to update those values in the code so that you don't need to run calibration every startup.

scottbez1, 2022-04-16:
Closing to keep the issues list clean. Please feel free to comment or open a new issue if your question hasn't been resolved.

closed 2022-04-16 by scottbez1