scottbez1/smartknob #63: Discord server

Paulsvcrch, 2022-04-14:
Taking into account the number of people interested, the need of help for few one and all the discussions especially about the motor, a Discord server could be a good thing don't you think so ?
Building a small community in order to develop/improve and help each other

I'm not a discord pro but I can try to help if needed

jsprenkle, 2022-04-14:
Does discord retain searchable content? If you can't find the answers via a
search engine they're a lot less useful.

paulsvc, 2022-04-15:
Yes it does, honestly there is a lot of useful communities (python arduino stm32 esp32 sumplefoc and so on)

LoganDark, 2022-04-17:
> Does discord retain searchable content? If you can't find the answers via a search engine they're a lot less useful.

Nope.

- Discord servers can't be discovered via search engines
- Discord servers can't be exposed publicly. You have to join with a Discord account, and only then can you use the server search

Guilded would be a much better alternative. Servers can be viewed & searched without having to join and without even having to use an account

flambert860, 2022-04-17:
They're discoverable in discord, it could be linked to the GitHub page.

LoganDark, 2022-04-17:
> They're discoverable in discord, it could be linked to the GitHub page.

Discovery is limited to servers with 1,000 members

scottbez1, 2022-05-04:
Thanks for the discussion. I'm definitely open to starting a discord (in fact, I started setting one up for this project a while ago, but just haven't opened up invites yet).

Pros:
- lower barrier to entry for casual discussion, etc (opening a GitHub issue can be intimidating compared just sending a chat message)
- probably easier for the community to offer help to one another than over GitHub Issues
- ability to discuss/share things that aren't "Issues", like sharing build photos, other related projects, cool YouTube videos, etc

Cons:
- possibility for development discussion to end up fragmented between GitHub and Discord (so you end up having to follow both to keep up)
- less searchable than GitHub issues
- More active moderation potentially required due to higher volume of content

I'll probably open up the server at some point, just need to finish setting it up (and tbh, I need to get a little more familiar with Discord in general; I've been using Slack for the Splitflap community but it seems like Discord is the way to go these days).

LoganDark, 2022-05-04:
> * More active moderation potentially required due to higher volume of content

You could lessen the load on yourself by finding a few trusted community members to moderate for you. I would nominate myself but the only interaction I've had is leaving like 3 comments...

flambert860, 2022-05-06:
@scottbez1 @LoganDark I can also help if anything is needed.

k1-801, 2022-05-09:
I agree on the need of a Discord server, as it would allow people interested in the project to elaborate minor issues much faster, but, again, it only makes sense if the people involved in decision-making of the project actually have the time to participate in said discussions.

mmlb, 2022-05-10:
Just a random gh-stargazer of the project but if you want chat and not slack, but find publicly searchable/readable logs important then zulip is a really great option: https://blog.zulip.com/2022/05/05/public-access-option/.

side note: If this discussion was locked away behind discord I would not have discovered it and would not have contributed.

LoganDark, 2022-05-10:
> zulip is a really great option: https://blog.zulip.com/2022/05/05/public-access-option/.

Oh that's nice! Keep in mind that literally happened 5 days ago, even my last reply was before that. But that's good to know!

mmlb, 2022-05-10:
> > zulip is a really great option: [blog.zulip.com/2022/05/05/public-access-option](https://blog.zulip.com/2022/05/05/public-access-option/).
> 
> Oh that's nice! Keep in mind that literally happened 5 days ago, even my last reply was before that. But that's good to know!

Definitely. Zulip was just another closed off garden until then and I personally hope more projects check it out vs discord.

scottbez1, 2022-10-03:
Oops, forgot to update this issue. Discord server now exists: https://discord.gg/5jyhjcbTnR

closed 2022-10-03 by scottbez1