scottbez1/smartknob #64: Idea: use a Raspberry pi 2W board

radu-boboc, 2022-04-14:
I know we got  bigger issues right now, with missing motors but I was thinking that if this project is to have success above the cool factor of the knob, then it needs to be able to do stuff like integrate into project's.
I understood from reading the issues that we don't have enough memeory available to import WiFi libraries.
So that means no connectivity to real projects.

So I propose to use instead the much more powerful Raspberry Pi 2W board. It has lots of memory, lot's of storage, good WiFi and Bluetooth connectivity and lots of IO pins.
There are two options integrate the board: directly by making a pinout holes on the PCB, or just solder the chip that can be bought separately.

What do you think of this migrate to Raspberry Pi proposition?

dylmeistah, 2022-04-21:
I feel like this could work, but a proof of concept would first be needed to call this a viable solution.

vlad-the-compiler, 2022-04-30:
I personally think this would be overkill for such a project, especially since it's an input device, and you shouldn't have to wait for a peripheral to boot a full-fledged Linux OS after you plug it in.

This is the type of project where the best fit is a microcontroller.

ghamauricio, 2022-04-30:
> I personally think this would be overkill for such a project, especially since it's an input device, and you shouldn't have to wait for a peripheral to boot a full-fledged Linux OS after you plug it in.
> 
> This is the type of project where the best fit is a microcontroller.

I totally agree.

Actually, I would really love to use this knob in my lighting controller project based on Arduinos and Arduino-compatible boards!

vlad-the-compiler, 2022-05-02:
> If anything the main board should be more focused, providing a USB HID and maybe a Bluetooth HID device, but no wifi

This sounds like the task for an nRF52840, which has Arduino support through their own Nano 33 BLE/Sense boards, or Seeed's small form factor 'Xiao nRF52' boards.
ESP32 support should still be a thing however, since a project like this is perfect for being adapted into a wall-mounted IoT controller.

radu-boboc, 2022-05-02:
> I personally think this would be overkill for such a project, especially since it's an input device, and you shouldn't have to wait for a peripheral to boot a full-fledged Linux OS after you plug it in.
> 

I also agree, it would be overkill and having a 10 second boot time to a cool button is not what we expect.
BUT!
We don't boot everyday and that 10 seconds could be once / year, and perhaps optimized a lot, by cutting down unnecessary stuff from the OS.

Also a Raspberry Pi would give the opportunity to do **stuff** with this button.
In the present configuration, with no WiFi, it is only a gimmick. Because of no connectivity, there is no other real world use other then a nice toy to show your friends.
If you want to integrate it with any of your projects, it should have WiFi + Bluetooth, plus space for some other software that we will write in the future.
As for cost, adding 10-15$ to a 100-200 button, is not so much. ( _actually less, because you will also remove the esp32_ ) and if we integrate it via SOC ( not board), it will cost even less.

I would argue that the benefits out-way the inconveniences so I keep my suggestion of making it useful and future proof.

ghamauricio, 2022-05-02:
> In the present configuration, with no WiFi, it is only a gimmick. Because of no connectivity, there is no other real world use other then a nice toy to show your friends.

Sure, hook it up to your Raspberry Pi using UART, SPI, I2C or any other communication method you want.

This is an HID device, and I agree it shouldn't be asked to do much more then HID stuff. If you want more functionality, add it yourself. That's what's expected from all of us who want to integrate such a neat little HID device into our projects.

vlad-the-compiler, 2022-05-03:
> Also a Raspberry Pi would give the opportunity to do stuff with this button.
In the present configuration, with no WiFi, it is only a gimmick. Because of no connectivity, there is no other real world use other then a nice toy to show your friends.

The current configuration is based on an ESP32, which has BT/BLE and Wi-Fi capabilities. It also is fast enough to provide a proper user interface on the display. It comes equipped with all of the interfaces you'd want, except for native USB - but that is supported on the S3 revision of the chip. Honestly, ESP32 is the perfect application processor for such a device, if we're to abide by the single responsibility principle.

ESP32 just makes a lot of sense for this particular project. A full-fledged computer? I fail to see a situation in which that would be needed. Nobody ever complains about their computer mouse having a 64MHz STM32 as their application processor, and for good reason.

k1-801, 2022-05-09:
Instead of a Raspberry Pi 2W, I'd suggest using a Raspberry Pi Compute Module, probably the latest cm4. The scenario where it could come in handy is a camera-based access control, with a camera (or two) mounted close to a corner of the outer case (two top corners in case of two cameras), and connected to the CM via CSI. However, in this scenario, the board will be under a heavy CPU load for image processing, and thus, a secondary microcontroller for basic motor/encoder interaction processing would be needed either way, as it has to be real-time processing. The display, on the other hand, could be connected to the RPI, the display is small enough to have little to none lag. Neither face recognition itself nor the decision-making have to happen on the CM, they could be delegated to the hub, but even transferring the images over wifi is heavy enough to consider having a secondary MCU solely dedicated to the haptic features.

scottbez1, 2022-10-03:
Cleaning up older issues. For reasons discussed (including https://github.com/scottbez1/smartknob/issues/64#issuecomment-1115502060), I don't think it will make sense to use to a full Linux system in place of the microcontroller, though we may want to change to a different microcontroller at some point.

closed 2022-10-03 by scottbez1