scottbez1/smartknob #66: Voltage regulator and logic shifter alternatives?

edwardrossi, 2022-04-16:
In my locale (Australia) Mouser has stock for all components bar two: U6 [AP2114HA-3.3TRG1](https://au.mouser.com/ProductDetail/621-AP2114HA-3.3TRG1) and U7 [SN74LV1T34DBVR](https://au.mouser.com/ProductDetail/595-SN74LV1T34DBVR).

These components look to be sold out globally with a lead time of at least two and nine months respectively.

I'm wondering whether [AP2115R5-2.5TRG1](https://au.mouser.com/ProductDetail/621-AP2115R5-2.5TRG1) might be a compatible replacement for [AP2114HA-3.3TRG1](https://au.mouser.com/ProductDetail/621-AP2114HA-3.3TRG1) as it looks like mouser has a 20 day lead time, as oppose to 9 months for [AP2114HA-3.3TRG1](https://au.mouser.com/ProductDetail/621-AP2114HA-3.3TRG1). The package and specs look identical.

scottbez1, 2022-04-17:
With the chip shortage, I'm not really seeing any good alternatives in the same package/pinout. If you're willing to modify the PCB design, you could look if you can find a 1117-style regulator (same size package, but different pinout). Or just find a suitable regulator >5V input, 3.3V output at >600mA and update the PCB for whatever package you can find in stock...

That AP2115R5-2.5TRG1 is not going to work though - it's a 2.5V regulator and also a different package (I didn't check size exactly but the pin spacing is different and it also requires connecting its EN pin); and the 3.3V version appears to be unavailable as well in any case...

If you can find AP7361C-33E-13, I believe that is a pin-compatible drop-in replacement, but also low availability...

Chaphasilor, 2022-04-26:
+1 on the Do Not Place. Not only does it offer more flexibility, it can also help with debugging and repairs, because it exposes pads :)