scottbez1/smartknob #67: Question: How to get detents/holding torque at positions between motor coils

aidanchandra, 2022-04-18:
Hi all!

We are a team at Stanford trying to replicate your results on a PCB-based motor for low-cost and lower-BOM count. 

The original motor demonstrated appears to have 14 coils (7 pole-pairs). In the video, the amount of detents far surpasses 14 individual possible detents. By detents, I mean the state where the knob is torqued to a single position (or angle). 

It's my intuition that it's trivial to hold the motor at one of the 14 coils (effectively getting detents at about 25degrees). You simply turn the correct coil on closest to the desired position once the encoder reads that it is close. 

How do you effectively provide a constant locked-torque at any arbitrary angle, despite there being only 14 coils - meaning only 14 positions where you can turn on the magnetic field? 

Best wishes

rtbcoop, 2022-04-18:
3-phase BLDCs usually use 6 step communications, so there are 6 steps per ELECTRICAL revolution.  A 14-pole motor will have 7 ELECTRICAL revolutions per MECHANICAL revolution (14/2).  Multiply those together, and you have 42 steps per MECHANICAL revolution or 8.6 degrees per step.  If you want finer steps, you'll need to increase your pole count or add some sort of gear reduction.

aidanchandra, 2022-04-18:
@rtbcoop Thank you so much for the answer.

What do you mean by 6-step communications and 6-steps per ELECTRICAL revolution?

rtbcoop, 2022-04-18:
Think of a single phase sinusoidal wave.  The wave starts at 0, goes up, comes back to 0 at 180 degrees, goes negative, and returns again to 0 at 360 degrees.  That's one electrical rotation.  When operating a BLDC motor, we typically have three phases, spaced at 120 degree increments.  Instead of a sinusoidal wave on each phase, we have DC pulses.  Three phases, each with one 0 and one + pulse make for 6 steps.  The linked article does a much better job of describing this control scheme.

https://www.mathworks.com/help/mcb/ref/sixstepcommutation.html

aidanchandra, 2022-04-19:
Ah - I had confused that with phases. We're still relatively new to all the jargon.

I _very_ much appreciate your help.

Closing the thread!

closed 2022-04-19 by aidanchandra