scottbez1/smartknob #71: Using off-axis sensor instead of on-axis

paulsvc, 2022-05-01:
Hi,

As you are not using the push button function of the sensor, maybe you should switch to an off-axis sensor such as the DRV5055.
It should allow more space and less complicated design maybe.

This is the solution I will try on my version

scottbez1, 2022-05-04:
Yeah. Theoretically the MT6701 could also just be mounted off axis and a LUT added to correct for the non-linearity. The LUT should be possible to generate completely automatically during initial calibration, since we don't _really_ care about the precise mechanical angle, just the relationship between electrical angle and mechanical angle. You potentially lose some resolution in some regions due to the non-linearity, but I'm not sure how much of an issue that would actually be.

teddosan, 2022-07-05:
For off-axis mounting, may I recommend the MA730 from MPS. Disclaimer: I'm the sensors FAE for MPS. However, I wouldn't be here recommending it if I didn't actually think it's a significantly improved solution here. Our MagAlpha parts contain a correction mechanism built into the IC so you don't have to build an external table. You can also implement push button detection functionality even off-axis.