scottbez1/smartknob #74: Debugging features : motor is not moving ...

shirogeek, 2022-05-08:
Hi there,

I finished assembly of everything and almost everything seemed to work on first boot (kind of an achievement knowing my track record).
This got me wondering if there is already or will be implemented debugging features which would allow you to single out faulty or unresponsive components. The assembly took me quite some time maybe because of my experience but such a feature would help accelerate debugging i would say.
Now specifically, in my case, the motor is unresponsive i feel no feedback at all. I am using an alternative : https://a.aliexpress.com/_mqVGhfy. It's a 12V rated brushless DC but from uni memory you can drive a motor irrelevant of voltage as long as current is enough to bypass cogging when static. I also have two others I haven't tried :
https://a.aliexpress.com/_mLbmAUa and
https://a.aliexpress.com/_mrdkyyE. Any advice on how I got about debugging this ? Are these motors even compatible to begin with ? Could I be guided into one or two lines of code I can add in the main.cpp to just make the motors rotate right or lzft at constant velocity (sanity check that the TMC chip works well) ?

Thanks a lot !

shirogeek, 2022-05-14:
I went ahead and did more tests today as I have again free time. I learned going through the issues that you can launch a calibration (motor specific) procedure by pressing Y at boot (thought it was for pressure sensing initially) and I can confirm something's wrong : 

`Sensor measures positive for positive motor rotation: NO, Direction=CCW
HX711 reading: 576402
millilux: 43970.40
Going to measure 20 electrical revolutions...
Going to electrical zero...
pause...
HX711 reading: 579336
millilux: 43632.00
Measuring...
HX711 reading: 577822
millilux: 43732.80
HX711 reading: 576790
millilux: 43632.00
HX711 reading: 577252
millilux: 43732.80
HX711 reading: 577066
millilux: 43156.80
HX711 reading: 576430
millilux: 43336.80
Electrical angle / mechanical angle (i.e. pole pairs) = -263404.19
Pole pairs set to -263404
HX711 reading: 578487
millilux: 42508.80
HX711 reading: 578432
millilux: 42076.80
1.11, 80.30, 79.19

RESULTS:
  zero electric angle: 6.09
  direction: CCW
  pole pairs: -263404
`

This is the result of the calibration. I monitored the total current going to the system and it never changes. I assume therefore that there is a problem with my electronics ? 
Next debugging steps I will look at U V W signals with my oscilloscope but just to clarify things, if I run the calibration the motor is expected to rotate correct ?

shirogeek, 2022-05-14:
Edit : I just realized a post already exists for this : https://github.com/scottbez1/smartknob/issues/36.
Sorry for duplicate... I will only reopen after steps in above issue are replicated.

closed 2022-05-14 by shirogeek

shirogeek, 2022-05-14:
I managed to make it work...

Basically I debugged the signals and observed I wasn't getting any output at U V W or got only at V and W and nothing at U basically weird shaenenigans. My problem was that I was using low temp paste when I soldered the entire board. Being a very small package, low temp solder tends to not "draw" the chips to their proper alignment. I switched to regular temp solder paste and got it working on the first try. It is indeed a tricky chip to solder as scott mentioned.

Now I move on to investigating several motors and the possibility to use an HDD motor as drive (apparently has the lowest cogging possible). I will report my progress...

ibrahimballoul, 2026-09-11:
@shirogeek  hi, i am bit late LOL, did you try the hdd motor?