scottbez1/smartknob #76: Firmware flashing

bulior, 2022-05-20:
Can anyone help me to flash the firmware. The documentation is still in progress.
I tryed to upload with VS Code and PlatformIO but the board is in boot loop.

With my test code to check the DMS it is possible to run the system:
```
/*
  Name:    Smartknob_test.ino
  Created: 5/19/2022 11:51:51 PM
  Author:  bulior
*/

//  LED´s
#include <FastLED.h>
#define NUM_LEDS 8
#define PIN_LED_DATA = 7

//  Note: Allowed to use pin 7 in FastLED lib
//  #define FASTLED_UNUSABLE_PIN_MASK (0ULL | _FL_BIT(6) | _FL_BIT(8) | _FL_BIT(9) | _FL_BIT(10) | _FL_BIT(20))
CRGB leds[NUM_LEDS];

//
//using namespace std;
//#include <algorithm>
#include <HX711.h>
HX711 scale;
uint8_t dataPin = 38;
uint8_t clockPin = 2;
long press_value_unit;

float hue;

//Display
//#include "roboto_light_60.h"

// the setup function runs once when you press reset or power the board
void setup() {
  delay(250);
  Serial.begin(19200);
  Serial.println("Smart Knob BEN 2022-04-23");
  
  FastLED.addLeds<SK6812, 7, GRB>(leds, NUM_LEDS);  //  LEDs
  for (int i = 0; i < NUM_LEDS; i++) {
    leds[i] = CRGB::Blue;
  }
  FastLED.setBrightness(50);
  FastLED.show();
  
  scale.begin(dataPin, clockPin);                   //  DMS

}

// the loop function runs over and over again until power down or reset
void loop() {
  //Serial.println("LED");
  delay(50);
  hx711read();

}

void hx711read() {
  const int32_t lower = -950000;
  const int32_t upper = -360000;

  if (scale.wait_ready_timeout(50)) {
    int32_t reading = scale.read();
    Serial.println(reading);
    // Ignore readings that are way out of expected bounds
    if (reading >= lower - (upper - lower) && reading < upper + (upper - lower) * 2) {
      static uint32_t last_reading_display;
      if (millis() - last_reading_display > 1000) {
        Serial.print("HX711 reading: ");
        Serial.println(reading);
        last_reading_display = millis();
      }
      long value = constrain(reading, lower, upper);
      Serial.print("Wert: ");
      Serial.print(value);
      press_value_unit = 1. * (value - lower) / (upper - lower);
      Serial.print(" / ");
      Serial.print(press_value_unit);
      hue = (value-upper)*-1.00;
      hue = hue/586000.00;
      hue = hue*255.00;
      Serial.print(" / ");
      Serial.println(hue);
      
      for (int i = 0; i < NUM_LEDS; i++) {
        leds[i] = CHSV(hue, 255, 255);
      }
      FastLED.show();
      
    }
  }
  else {
    Serial.println("HX711 not found.");

    for (uint8_t i = 0; i < NUM_LEDS; i++) {
      leds[i] = CRGB::Green;
    }
    FastLED.show();

  }
}
```

AR17HY, 2022-05-27:
You should be able to get out of boot loop by using the boot push button on the esp32 board :
1. press BOOT switch (maintain it)
2. press EN switch (and release it)
3. wait 1 sec and release the boot switch
You should be able to access the MCU through the UART/USB port (seen as COM port on your windows machine) and flashing must be all right.

If it does not work : 
Can you share the platformio.ini ?
What OS are you running VSC ?
Can you confirm the hardware you are using ? (esp32 version and USB driver)

scottbez1, 2022-05-28:
Can you share the debug output from the boot loop when you use Upload and Monitor? That will help pin down exactly what's happening. One known issue is that you'll get a bootloop if you don't have the VEML7700 ALS sensor - the I2C initialization causes a bootloop if it's not present. So if the error message is related to I2C, I'd recommend turning off [SK_ALS ](https://github.com/scottbez1/smartknob/blob/0b38eac75d5aec026bdfcf168778379b9c4865fa/firmware/platformio.ini#L53) and try again. Otherwise, hopefully sharing the log output will help identify the issue.

alex934, 2022-09-22:
> You should be able to get out of boot loop by using the boot push button on the esp32 board :
>
> 1. press BOOT switch (maintain it)
> 2. press EN switch (and release it)
> 3. wait 1 sec and release the [...]

What push button are you referring to? There are no push button on the T-micro 32
[image]

bulior, 2023-02-28:
Okay thanks i fix the boot loop by resoldering and i'm able to flash my firmware.

But if i try to flash your general firmware i will have some issues.
I implemented your files as project in Visual Studio Code and compiled with [env:view] profile. 
All librarys has been downloaded but i will get lots of error messages

`Processing view (board: esp32doit-devkit-v1; platform: espressif32@3.4; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (3.4.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6) 
 - tool-esptoolpy @ 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Simple FOC @ 2.2.0
|   |-- Wire @ 1.0.1
|   |-- SPI @ 1.0
|-- TLV493D-Magnetic-Sensor @ 1.0.3
|   |-- Wire @ 1.0.1
|-- AceButton @ 1.9.1
|-- TFT_eSPI @ 2.4.25
|   |-- SPI @ 1.0
|   |-- SPIFFS @ 1.0
|   |   |-- FS @ 1.0
|   |-- FS @ 1.0
|-- FastLED @ 3.5.0
|   |-- SPI @ 1.0
|-- HX711 @ 0.7.5
|-- Adafruit VEML7700 Library @ 1.1.1
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 1.0.1
|   |   |-- SPI @ 1.0
|   |-- Wire @ 1.0.1
|   |-- SPI @ 1.0
Building in release mode
Compiling .pio\build\view\libfe7\FastLED\FastLED.cpp.o
Compiling .pio\build\view\libfe7\FastLED\bitswap.cpp.o
Compiling .pio\build\view\libfe7\FastLED\colorpalettes.cpp.o
Compiling .pio\build\view\libfe7\FastLED\colorutils.cpp.o
Compiling .pio\build\view\libfe7\FastLED\hsv2rgb.cpp.o
Compiling .pio\build\view\libfe7\FastLED\lib8tion.cpp.o
Compiling .pio\build\view\libfe7\FastLED\noise.cpp.o
Compiling .pio\build\view\libfe7\FastLED\platforms.cpp.o
Compiling .pio\build\view\libfe7\FastLED\platforms\esp\32\clockless_rmt_esp32.cpp.o
Compiling .pio\build\view\libfe7\FastLED\power_mgt.cpp.o
Compiling .pio\build\view\libfe7\FastLED\wiring.cpp.o
Compiling .pio\build\view\src\display_task.cpp.o
.pio/libdeps/view/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::init(gpio_num_t)':
.pio/libdeps/view/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:111:15: warning: variable 'espErr' set but not used [-Wunused-but-set-variable]
     esp_err_t espErr = ESP_OK;
               ^
.pio/libdeps/view/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: In member function 'void ESP32RMTController::startOnChannel(int)':
.pio/libdeps/view/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:239:15: warning: variable 'espErr' set but not used [-Wunused-but-set-variable]
     esp_err_t espErr = ESP_OK;
               ^
Compiling .pio\build\view\src\interface_task.cpp.o
Compiling .pio\build\view\src\main.cpp.o
Compiling .pio\build\view\src\motor_task.cpp.o
Compiling .pio\build\view\src\mt6701_sensor.cpp.o
Compiling .pio\build\view\src\proto_gen\smartknob.pb.c.o
In file included from src/proto_gen/smartknob.pb.c:4:0:
src/proto_gen/smartknob.pb.h:6:16: fatal error: pb.h: No such file or directory
[... 60 lines cut ...]
************************************************************

compilation terminated.
*** [.pio\build\view\src\interface_task.cpp.o] Error 1
In file included from src/motor_task.h:8:0,
                 from src/motor_task.cpp:3:
src/proto_gen/smartknob.pb.h:6:16: fatal error: pb.h: No such file or directory

************************************************************
* Looking for pb.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pb.h"
* Web  > https://registry.platformio.org/search?q=header:pb.h
*
************************************************************

compilation terminated.
*** [.pio\build\view\src\motor_task.cpp.o] Error 1
================================================================================= [FAILED] Took 29.18 seconds =================================================================================

Environment    Status    Duration
-------------  --------  ------------
view           FAILED    00:00:29.180
============================================================================ 1 failed, 0 succeeded in 00:00:29.180 ============================================================================ 
`