scottbez1/smartknob #78: Mitigate LCD Tearing

sungdzda, 2022-06-07:
First of all many thanks for sharing the project.

As is known LCD without tearing mitigation/Vsync is good for doing small-scale local updates, but uncool for global updates and unfit for playing animations. The GC9A01 does not offer double-buffered GRAM nor is the TE pin accessible on that particular display module.

This is not me nitpicking on the project but rather what I see as an important differentiation between a DIY hobby grade project and something refined and polished with potential for large scale adoption for the benefit and enjoyment of other people. Along the same lines when you go through the project you can find many unholy DFM/DFA issues. Those expensive nylon prints and the rather costly TMC driver chip and laborious installation of the strain gauges are examples.

Any ideas would be appreciated.

scottbez1, 2022-06-07:
Thanks for the feedback. Would love to find a cheap round LCD with TE available, but so far I have not seen any such display readily available, so we make do with what's available on the market today.

Regarding the "unholy DFM/DFA issues" - while these are valid points, it's important to clarify that this is the first generation of a hobby-grade project, meant for advanced hobbyists to assemble by hand; this is not intended for mass production, so certain decisions were made to accelerate development at the expense of mass-manufacturability. Not to say this can't improve, but it was a conscious decision to make this tradeoff (e.g. hand-wiring the display is a pain, but the alternative of having custom ribbon/flex cables manufactured would have been much more expensive and more complicated/slower to design).

The "expensive nylon prints" are entirely optional - you can FDM 3d print the parts for this (this is how I validated the design in the first place), but they just won't look as nice, so I chose to pay to order MJF prints. There's not really any alternatives I'm aware of that would be cheaper, other than tooling injection molds, which is well beyond the scope of this project.

I'm not sure I understand what you mean by "rather costly TMC driver chip" - it's $2.50 at single quantities (and one of the few, if only, integrated-mosfet BLDC drivers available in this voltage range). Recommendations for alternatives would be appreciated if you have them!

Laborious installation of the strain gauges is definitely on the radar. When I have some free time (currently all of my effort is going into [finding and qualifying motors](https://github.com/scottbez1/smartknob/issues/16#issuecomment-1148311100)), I'm planning to experiment with the SMD resistor technique described here: https://github.com/scottbez1/smartknob/issues/24#issuecomment-1073540944

sungdzda, 2022-06-07:
Hey, thanks for the reply.

What is the merit offered by the TMC chip, if you are already generating full-fledged 6-ch PWM signals from a processor? It's not like the chip has any in-built smarts _at all_. Going discrete is the intuitive choice here for an optimized BOM cost. Putting it in perspective, the TMC is almost as expensive as all the other PCB components combined. (Excl. the Lilygo module, which is overpriced in and of itself.) It is about 2x as expensive as an ESP32 processor. For over-current protection, simply prepending the circuit with a self-protected Si load switch or PPTC fuse would suffice, because the motor's current requirement is so low.

thalesfsp, 2022-06-17:
@sungdzda Why don't you contribute to the project? Apparently you have a lot of useful knowledge..

scottbez1, 2022-10-03:
Cleaning up old issues. Feel free to reopen any specific, focused issues that are actionable.