scottbez1/smartknob #81: Why bldc?

MahOraibi, 2022-07-01:
Hello,

This is not exactly an issue but a question. Since this is a knob and requires (accurate steps) and absolute smoothness is not a requirement why did you choose bldc over a small stepper motor like nema14 ? Is it because of the shaft? Because these steppers are pretty small, readily available everywhere, very easy to program and operate.

h04x3d, 2022-07-21:
Hi,
To the best of my knowledge, brushless motors can provide more smooth and controlled rotation and also don't impact the motion of the encoder when detents are disengaged which a stepper motor can't provide.
Hope this helps.

5H4D0W-X, 2022-07-31:
Hello,
There are several reasons that a BLDC is better in this context. For good haptic feedback we need to control the force applied by the motor, and that's easier to do with a BLDC than with a stepper. In addition a stepper has very pronounced cogging and not really a way around it due to the internal structure and working principle. Steppers have way more "physical" holding torque (again, due to the internal structure) so any detent would make it very difficult to turn without heavy current limiting, a BLDC always has some spring and backdrivability. Basically, a BLDC can produce more "organic" and smoother motion which is very important when you want to use it for haptics

MahOraibi, 2022-07-31:
Thank you both for giving a very detailed answer. This is really helpful.  Thanks very much. It's very clear now

closed 2022-07-31 by MahOraibi