scottbez1/smartknob #82: can this be used

drraynr, 2022-07-04:
https://www.aliexpress.com/item/1005001968864724.html?spm=a2g0o.home.15002.2.650c21450yDWtU&gps-id=pcJustForYou&scm=1007.13562.275843.0&scm_id=1007.13562.275843.0&scm-url=1007.13562.275843.0&pvid=246253d0-0f66-4f97-b942-8d9fc2c229e1&_t=gps-id:pcJustForYou,scm-url:1007.13562.275843.0,pvid:246253d0-0f66-4f97-b942-8d9fc2c229e1,tpp_buckets:668%232846%238111%231996&pdp_ext_f=%7B%22sku_id%22%3A%2212000018267338478%22%2C%22sceneId%22%3A%223562%22%7D&pdp_npi=2%40dis%21INR%21%21219.76%21%21%21%21%21%4021032fa116569185167104436eb4e7%2112000018267338478%21rec

tinkrmind, 2022-08-19:
Found another source: https://www.aliexpress.com/item/3256802907900422.html?spm=a2g0o.detail.1000014.7.3716240aGpPV5p&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40050.281175.0&scm_id=1007.40050.281175.0&scm-url=1007.40050.281175.0&pvid=723e6ab3-f152-4f12-a1bd-a474cf569f22&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40050.281175.0,pvid:723e6ab3-f152-4f12-a1bd-a474cf569f22,tpp_buckets:668%232846%238109%231935&pdp_ext_f=%7B%22sku_id%22%3A%2212000024047876655%22%2C%22sceneId%22%3A%2230050%22%7D&pdp_npi=2%40dis%21USD%213.1%212.39%21%21%21%21%21%40210312ee16609189296486113e90e3%2112000024047876655%21rec

scottbez1, 2022-08-19:
That motor has too much cogging and is therefore not a good fit. Please see https://github.com/scottbez1/smartknob/wiki/Motor-Status

closed 2022-08-19 by scottbez1