scottbez1/smartknob #87: 关于VEML7700的bug

a842690807, 2022-08-10:
提交个bug，关于VEML7700的，不是很严谨，没有多次验证。
如果不焊接VEML7700，下载完代码后，整体程序会处于一直重启的状态。
日志如下：

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
ALS sensor not found!
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400f844e  PS      : 0x00060b30  A0      : 0x800d9f79  A1      : 0x3ffb2ef0  
A2      : 0x00000000  A3      : 0x00000011  A4      : 0x0000000f  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb2ed0  
A10     : 0x00000000  A11     : 0x3ffb827c  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060b20  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000e  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400f844e:0x3ffb2ef0 0x400d9f76:0x3ffb2f10 0x400d2a82:0x3ffb2f30 0x400d2cf9:0x3ffb2f90 0x400883f2:0x3ffb2fb0

Rebooting...

以上现象是在我偷懒的时候，当我焊接完成了其他器件而没有焊接VML7700时，代码运行到ALS sensor处时一直重启。
我尝试拆除了几个主要芯片后不能得到改善，然后我就把VML7700焊上后，代码就可以正常运行了。
这个具体的我还没看代码实现，我认为也许是初始化未成功后继续调用导致的错误。
以上供大家参考

ekin1106, 2022-08-29:
platformio.ini 里-DSK_ALS=0 就可以跳过veml7700

a842690807, 2022-08-29:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

scottbez1, 2022-10-03:
https://github.com/scottbez1/smartknob/issues/87#issuecomment-1230189698 is correct,`-DSK_ALS=0` will disable the ALS

closed 2022-10-03 by scottbez1