scottbez1/smartknob #94: What is the tlv_sensor.h meaning？

L-SY, 2022-09-23:
I see tlv_sensor.h in you project it looks like for Tlv493d,
but I do not see Tlv493d in your PCB.So what is  tlv_sensor.h meaning?

Could you teach me ? Thanks

seuf12, 2022-09-23:
It's just a library for the magnetic encoder chip

scottbez1, 2022-09-24:
In earlier versions I experimented with using the TLV493d magnetic sensor. It's not a very good chip though, so the MT6701 is the current recommendation instead. That file is leftover from that, and would allow you to use a TLV493d if you wanted to.

closed 2022-09-24 by scottbez1

wchill, 2024-04-06:
Seems like there's a newer version of the TLV493D that has the bug fixed: https://community.infineon.com/t5/3D-Hall-Magnetic-sensor/TLV493-A1B6-lockup-crash-bug/m-p/346096#M154

Might be worth looking into