scottbez1/smartknob #111: Serial Monitor showing gibberish data

pipipipi2002, 2023-02-26:
I managed to flash the firmware into the ESP32, but when im monitoring the serial, it gives me gibberish data.
[image]

I cloned the code directly from github and uploaded, according to the guide provided by scott. Thank you in advance!

staalebk, 2023-03-08:
Did you set the correct baud rate? I think the default for this project is 921 600.

ChristianEmeis, 2023-04-02:
Might be a little bit late, but from your screenshot, I can see that your serial monitor is running at the correct default baud rate of 921600. From 2 Smartknobs I have one with a similar issue, changing the baud rate to 115200 fixed it for me, but I'm not sure what’s the root cause of this problem, as it worked with 921600 before. To change the baud rate, go to your platform.ini file and under base_config change monitor_speed and DMONITOR_SPEED under build_flags to your desired baud rate.

pipipipi2002, 2023-04-07:
Hmm that seems to have solved the problem. Thank you very much.

closed 2023-04-07 by pipipipi2002