scottbez1/smartknob #126: Calibration problem

gregory353a, 2023-06-24:
During calibration motor won't start rotate without little push. It often fails whole calibration.

`Starting calibration, please DO NOT TOUCH MOTOR until complete!
LOG: HX711 reading: 1073822
LOG: millilux: 741.60
LOG: HX711 reading: 1074068
LOG: millilux: 741.60
LOG: HX711 reading: 1074621
LOG: Changing config to 0 -- Unbounded
No detents
LOG: Changing config to 1 -- Bounded 0-10
No detents
LOG: Changing config to 2 -- Multi-rev
No detents
LOG: Changing config to 3 -- On/off
Strong detent
LOG: millilux: 734.40
LOG: HX711 reading: 1077951
LOG: millilux: 734.40
LOG: 
LOG: Sensor measures positive for positive motor rotation:
LOG: YES, Direction=CW
LOG: HX711 reading: 1075398
LOG: millilux: 734.40
LOG:   (start was 9.7, end was 11.5)
LOG: Going to measure 20 electrical revolutions...
LOG: Going to electrical zero...
LOG: HX711 reading: 1075476
LOG: pause...
LOG: millilux: 734.40
LOG: HX711 reading: 1075424
LOG: Measuring...
LOG: millilux: 734.40
LOG: HX711 reading: 1075550
LOG: millilux: 741.60
LOG: HX711 reading: 1075442
LOG: millilux: 734.40
LOG: HX711 reading: 1075107
LOG: millilux: 2124.00
LOG: HX711 reading: 1059738
LOG: millilux: 1116.00
LOG: HX711 reading: 146598
STATE: -1 [-6, 6]  (detent strength: 1.00, width: 60 deg, endstop strength: 1.00)
LOG: Electrical angle / mechanical angle (i.e. pole pairs) = -5675.98
LOG: ERROR! Unexpected calculated pole pairs: -5675.98
LOG: applying position change
LOG: adjusting detent center
LOG: Got new config
LOG: applying position change
LOG: adjusting detent center
LOG: Got new config
LOG: applying position change`

gregory353a, 2023-06-24:
```
Starting calibration, please DO NOT TOUCH MOTOR until complete!
LOG: millilux: 403.20
LOG: HX711 reading: 1075288
LOG: millilux: 583.20
LOG: HX711 reading: 1075457
LOG: millilux: 576.00
LOG: HX711 reading: 1075472
LOG: millilux: 583.20
LOG: HX711 reading: 1076203
LOG: 
LOG: ERROR! Unexpected sensor change: start=10.56 end=10.56
```

jseyvecou, 2023-07-04:
Having the exact same issue, did you find any fix by any chance?

badaaim, 2023-07-14:
I am trying different motors to the one used in the project and I get similar weird values for pole pairs. Not sure if yours is the same case but at least for me if I switch back to the motor used in the project then I get more normal values like 7 for pole pairs.

gregory353a, 2023-07-16:
> Having the exact same issue, did you find any fix by any chance? I tried replacing some components like the MT6701-CT with a new one, but the issue persists. The motor I use is the one from SparkFun linked in the other thread

Unfortunately I didn't find problem yet. I will investigate, but I need some free time for that ;)

Jirik-Cermak, 2024-10-02:
> I am trying different motors to the one used in the project and I get similar weird values for pole pairs. Not sure if yours is the same case but at least for me if I switch back to the motor used in the project then I get more normal values like 7 for pole pairs.

Hello, I have the same issue. I see also 7 poles. What is a problem for me is, in two quadrants is encoder response to rotation higher and in two quadrant is the response less than reality. 

I a good idea solve this by interpolation table? Do You have better idea?

Thank You
Jiri

xKraig, 2024-10-10:
> I have the same issue. The first calibration ran fine, but ever since I am getting either the "Unexpected sensor change" or "Unexpected calculated pole pairs" error.

To follow up on this, I fixed it by replacing the motor.