scottbez1/smartknob #131: Motor continuously spins or feels jerky when stationary

jtrexr2, 2023-08-03:
I created a version of this project that uses the same motor from Sparkfun, TMC6300, MT6701, HX711, etc. The difference is that I'm using an ESP32-S3-DevKitC-v1.0, but I've updated the platformIO project and am able to compile/run code. However, with any of the configs that include detents, the motor either continuously spins or feels and looks "twitchy" while stationary. I assume this is a PID/torque issue, but I feel like my setup is similar enough to yours that the code should produce similar smoothness that I see in your videos. Any ideas?

wandsoning, 2023-08-20:
https://item.taobao.com/item.htm?spm=a1z10.3-c.w4002-24948780880.33.237490f1oyBnyd&id=724766670182
 you can buy the motor 3215

xKraig, 2024-07-02:
have you tried calibrating the motors via the inbuilt calibration function?