scottbez1/smartknob #140: simplyfoc library issue

wangxinghe88, 2023-10-16:
Askuric's simplyfoc project has been updated to 2.3.1 and there is changes in function's parameteres,  which will generate error in the code.
solution：
In the platfromio.ini file, lib_deps should be changed to:
askuric/Simple FOC@2.3.0

I think the typo "@^2.3.0" makes the version up to V2.3.1 unintentedly.

Edo78, 2023-10-16:
I think the issue is in Askuric's simplyfoc ... if a breaking changes is introduced (like a change in function's parameters) then he should have followed semantic version rules and not just bump a patch

closed 2023-10-18 by scottbez1

runger1101001, 2023-10-19:
Yeah, sorry about that!

The changes were documented and announced, but we should pay attention to the version numbers as well.

I created a PR: https://github.com/scottbez1/smartknob/pull/142
this fixes the compile errors, but as I don't have a smartknob I can't test it out...

PR #141 (2023-10-18): peg sfoc version
nanofoc simpleFOC version peg at 2.3.0

Fixes #140
commit dafc5ca1 (2023-10-18): peg sfoc version (#141)

nanofoc simpleFOC version peg at 2.3.0

Fixes #140