scottbez1/smartknob #146: Esp32 Boot Pin keeps oscillating between 3.3 and 0V

pipipipi2002, 2023-11-07:
When Im trying to flash the firmware to the board (newly soldered), im stuck at the "connecting" phase and failed after timeout. After I monitored the Boot pin, it seems to be oscillating between 3.3 and 0v. Any ideas why this is happenning? is it faulty CH340C chip?

thatguyshade, 2023-11-07:
Try flashing one more time with flash mode “dout” and flash size 4mb

pipipipi2002, 2023-11-07:
@thatguyshade Do you mean adding the -DOUT in the build flag and the flash size in the platform.ini config file?

pipipipi2002, 2023-11-07:
After some digging, I got what you meant. Set this in platform.ini and it still doesnt work.
```
board_build.flash_mode = dout
board_build.flash_size = 4MB
```

FYI this is just the baseboard with no motors/screen connected.

Oliver1948, 2024-10-06:
Having the same issue. Ever get it fixed?