scottbez1/smartknob #147: Interested in purchase of finished product

elzii, 2023-11-08:
Hello,

I work with the R&D department at a large automative company. We'd love to test one of these out. If you're interested in selling a preassembled model, please reach out to zizzo.alexander at gmail.com and I can provide my work email (did not want to disclose it here).

Excellent work btw. I had this mocked up in software ages ago, glad to see someone went and did it!

Chaphasilor, 2023-11-08:
Just wanted to point out that in case your company deems this as usable, it would be great if the company would contribute to the open-source project (or a fork thereof). After all, the project helped you out :)

elzii, 2023-12-30:
Oh absolutely they will be credited if it makes it to production.  An open source component in a production vehicle? Pure joy at the thought.

denishessberger, 2024-08-06:
We are also currently actively pursuing this for a project. I'll make sure to send you an email!

And9686, 2024-11-12:
> Just wanted to point out that in case your company deems this as usable, it would be great if the company would contribute to the open-source project (or a fork thereof). After all, the project helped you out :)

Not really relating to the title of this issue - A company contributing to the open-source project (?), don't you think a company would try to patent this 'force feedback implementation' and no one else could sell this again like nanofoc or seedlabs?

elzii, 2024-11-13:
> Not really relating to the title of this issue - A company contributing to the open-source project (?), don't you think a company would try to patent this 'force feedback implementation' and no one else could sell this again like nanofoc or seedlabs?

Patents exist already for this device held by Alphabet (Nest dials), and several industrial machinery manufacturers (seen these units on water jet cutters, C&C machines & more). 

No interest in patent -- or preventing sale to/by others.

adam-securafort, 2026-10-20:
gmail 😆