scottbez1/smartknob #148: Many improvements (battery, accelerometer, touch, MIDI, etc.)

DerSpatz, 2023-11-19:
First of all, I want to say that I really like the concept of the SmartKnob. That's why I decided to look into it more thoroughly and see where I can maybe add some improvements, some of which where also asked for here in the Github Issues. I am always trying to get the most out of a design, some might call it "feature creep".

# General Changes

First, I decided to go with a 4-layer PCB, where both inner layers are consecutive ground planes (plus some 5V fill on one layer). The top and bottom layers are signal, with a 3.3V fill. This improves signal integrity and EMI, and also makes routing (with good grounding) easier when space is limited, while the price is not that much higher than a 2-layer PCB.

Next, I moved the ESP32 to the bottom layer. This way, the ESP32 can be moved closer to the center of the PCB, as it does not collide with the outer ring of the top shell there. Now there's also enough space for an ESP32 in the WROOM form factor, so I went for the ESP32-S3, as it is faster, and has native USB, which might be useful if we want to use the SmartKnob as HID input device. My redesign is also ready to be used with a ESP32-S3-WROOM-2, which offers even faster flash and PSRAM. You just need to close one solder jumper for the WROOM-2. You can also use the "U" variant of the WROOM with an external antenna, which is also part of the PCB set.
I consider the WROOM module hand solderable, if you add a small plated slot to solder the center ground pads (used only for heatsinking). I also made sure that all parts on the back of the PCB are easy to hand solder, as are most of the top parts (min size 0603, enough distance between parts), except those darn small packages with bottom pads...

Then I cut down the corners of the PCB. You can still use a shell that has the shape of the original smart knob, but you can now also mount the SmartKnob in a round shell with 89mm or 84mm (U variant) diameter. The extra space can also be used for batteries (more on that later).
I also changed the PCB color to green, with a white silk screen ring where you can see it. The reason for this is simple: white 4-layer PCBs are much more expensive at JLCPCB, compared to green 4-layer.

Next, I made some mechanical changes: I moved the mounting holes on the PCB, as they where in a spot that was very tight, and I could not fit the necessary traces there otherwise. We also need a bit more space behind the PCB to accomodate some higher parts, but making the overall design 3-5mm higher should be enough. Please note that, while I considered and checked the feasibility of the mechanical changes, I did not do a mechanical redesign, yet (I'm not that good at 3D modeling).

At last, I added a PTC fuse for safety and TVS diodes to protect the USB and I2C lines. I also switched some parts:
The LDO voltage regulator was replaced with a DC-DC converter ([Texas Instruments TLV62569](https://www.ti.com/lit/ds/symlink/tlv62569.pdf)) for better efficiency.
The SK6812-SIDE-A LEDs where replaced with [WS2812-4020](https://www.peace-corp.co.jp/data/WS2812B-4020_V1.2_EN.pdf) LEDs. They are cheaper, have a built in capacitor (less work), and work with signal voltages down to 2.7V, so a level shifter is not needed anymore. _By the way, you can also power the first LED through a diode, the voltage drop works like a level shifter._
The light sensor was replaced with an THT analog version (looks like a LED). This allows for a wider variety in placement and parts choice. You could also mount the sensor through an opening in the top shell. The exact part number depends on your requirements in size (3mm or 5mm) and color (transparent or dark).

In the end, when all routing is done, I will also check where I can add some low height (max 6mm) bulk capacitors to the bottom side of the board for both power rails.

# New features:

Now that we have more space on the PCB and inside the case, we can start adding features...

## Accelerometer

The [ST LIS3DH](https://www.st.com/resource/en/datasheet/lis3dh.pdf) is a three axis accelerometer with interrupt output and ADC inputs. One ADC is used to read out the brightness sensor, the other ADC is used to read out the battery voltage.
With a built-in accelerometer (and USB support), turning the SmartKnob into a 3D mouse is much easier. My idea was to place the SmartKnob in a round case (hence the size reduction of the PCB) with a spherical bottom, which is then placed in a square baseplate with a mechanical design that allows movement of the SmartKnob while also keeping it centered. All necessary cables could be routed through a hole in the bottom.

## Battery Power

For battery power, I have designed an extra PCB around the [IP5306-I2C](http://www.injoinic.com/wwwroot/uploads/files/20200221/0405f23c247a34d3990ae100c8b20a27.pdf), which is an [I2C controlled](https://sharvielectronics.com/wp-content/uploads/2021/07/IP5306-I2C-registers.pdf) version of the IP5306. It charges a 3.7V lithium battery with 2.4A, while also delivering 5V/2.1A as a single chip solution. The power PCB is connected to the main PCB with two 5p 1.25mm cables (Molex PicoBlade). One is for power (5V, BAT, GND), and the other one transmits USB and I2C, so you can mount the power PCB into the bottom shell of the 3D SmartKnob and still use USB, without having a visible USB cable connected to the movable part of the 3D SmartKnob.

Please note that the batteries should not be too small, as they need to handle the charge and discharge current properly. Anything above 2000mAh should be enough. For the non-3D SmartKnob, you could add a flat (4-5mm) LiPo battery behind the main PCB. For the 3D SmartKnob, you could place up to 8 801350 LiPo batteries around the edge of the base unit.

On the main PCB the power input both from direct USB and the power PCB is controlled by a P-MOSFET. The gate from the PMOS can be controlled by the MCU (to keep it powered on), by a power button (as backup solution, for programming) or, most likely by the

## Touch Sensor

The [TTP233H-HA6](https://datasheet.lcsc.com/lcsc/2109061130_Tontek-Design-Tech-TTP233H-BA6_C2890349.pdf) is a single touch sensor. It is powered directly from the battery, but it has a very low standby current of 3µA, which should allow for a long standby time (The standby current of the IP5306 is 50µA). The input pin of the sensor is connected to one of the screw holes for the motor (I assume the motor is made of metal). This way the whole motor acts as an electrode and should be sensitive enough to detect if a hand is closing in on the SmartKnob, even though the distances are rather large. The MCU can also read out the output of the touch sensor, even though I assume this might be unreliable when the board is powered up, due to stray electric fields. But it should at least work when powered off. As a last measure, there's a solder jumper that adds a 100pF X7R 1206 capacitor in series to the much smaller sensing capacitor. The 1206 capacitor is placed on the bendy part of the PCB (just like the strain resistors), so a press on the SmartKnob should cause a change in capacitance in the 1206 capacitor, triggering the touch sensor.

**_The only question is: How long is the boot time? If it is too long, touch activation might not be a good user experience..._**

## Buttons

Besides the power button, I also added four additional switches around the edges of the PCB for user functions. These switches are side activated, so the button design on the top shell must have some kind of hinge and lever thingy that toggles the switches. This allows for much more freedom regarding the placement of the buttons. There are also some additional LEDs so the buttons can be illuminated if the lever/plunger of the switch is printed in a semi transparent material, with a cutout in the button shell showing the illuminated symbol.

## Display Cable

For easier manufacturing, the display cable was replaced with a 10p FPC cable with 0.5mm pitch. This cable is 5.5mm wide and fits through the center hole of the motor. The 3D printed tube through the motor must be replaced by two half-circle shaped columns, so the FPC cable fits in between. The routing of the cable is a bit weird, but I had to make sure it works with 100mm length FPC cables, as this is the most common length. The two additional traces can be used for better grounding (signal integrity, EMI), or if everything else fails, could router the battery voltage and a signal trace to the display PCB, so the touch se [... cut ...] D, so you can even make those buttons go blinky blinky.

**_What do you think of the changes? Would you try my design?_** 

I have not yet fully routed the PCB, just critical spots for now... but if there's interest, I would finish the routing and publish it.

Here's an image of the front and back of the PCBs, as you can see, all boards are included in the design, and can be separated from each other. Over all size when ordered is 100x77mm, so it's below the 100x100mm threshold:

[image]
[image]

Truetech000, 2023-11-20:
Yes, just yes! +1 from me.

feurer98, 2023-11-21:
i have no version built yet.  but I would join in here.  so +1

radu-boboc, 2023-11-21:
yes, it is awesome, please refine it further.

ChristianEmeis, 2023-11-25:
This looks really good, I would love to try this design when its finished 👍🏻

TKaluza, 2023-11-26:
Maybe adding zigbee would be nice, see [esp-docs ](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/introduction.html)

DerSpatz, 2023-11-26:
@TKaluza The MIDI board is connected via UART, so using a ESP32-H2-WROOM module instead of the MIDI board should be possible, if you want ZigBee.

Design is mostly done, but I am still waiting for a word from @scottbez1 ...

florianL21, 2023-12-01:
I am just about to order all the parts but the esp32 is a major concern for me especially because of the missing support for native USB.
I would be happy to build and test this version of the electronics instead.
As for the required mechanical changes, I should be able to make the needed adjustments myself and am happy to contribute them once tested and working

DerSpatz, 2023-12-02:
@florianL21 Sounds good, let's get in contact!

Ich schicke dir nächste Woche mal ne Nachricht, bin übers Wochenende unterwegs...

radu-boboc, 2023-12-08:
All the improvements  you talked about, are they in a branch? 
I would like to try them

DerSpatz, 2023-12-13:
Not yet, sorry. This is not ready for full release yet...

DerSpatz, 2023-12-13:
@florianL21 Ich sehe gerade, dass ich auf GitHub keine DMs senden kann. Kannst du mir eine EMail senden an: bastian bei mohi.ng (jo, nigerianische Domain, aber kein Scam ;-) )

radu-boboc, 2023-12-14:
I am curious if you ever considered a raspberry pi zero 2w.
From what I understood, and correct me if I am wrong, at this moment the project doesn't have WiFi so it can't be integrated with anything useful.
I am not gonna use it via usb, so In order to have more hardware resources we need a raspberry pi chip or something with more resources.

DerSpatz, 2023-12-14:
A Pi Zero 2W (or any Raspberry Pi) is not a good use case for this. Raspberry Pi runs a real Linux OS which, compared to a MCU, means:
- longer boot times
- higher power draw
- worse real time capabilities (needed for sensing and PWM controlling the dial)
- not enough pins
- way to much compute power

The ESP32 does have Wifi (and Bluetooth), and the ESP32-S3 even can be used as an USB HID device (which the Pi can't). After all, this is just a very sophisticated input device, so it should not need much computing power, that can be done by the system that takes the inputs.

radu-boboc, 2023-12-15:
Agreed with what you said, but can we use it to control stuff arround house? Oe use it to control stuff in windows/linux?
Can it send commands via WiFi (not usb) to target devices?

nicolabortignon, 2023-12-22:
@DerSpatz since you mentioned, have you got in your revision the knob to be recognized as a USB (C) HID device ?

kawashimaken, 2025-01-11:
@DerSpatz  Amazing! Do you have a GitHub Repo for this?

adamNewell, 2025-08-04:
Hi @DerSpatz, even in an unfinished state, it would be awesome if you could publish what you have so far. I'm super interested in finishing your version out!