scottbez1/smartknob #150: Idea: Give option to buy pre-made PCBs somewhere for less experienced people

the-baaron, 2023-12-18:
Hi! I'm fascinated by this project. I've been following it for a while, and now the motor is back in stock I'd love to build one of these myself. 

However, I only have some basic soldering stuff, and not nearly enough experience to add the chips to the PCBs. It would be amazing if there was an option to buy the PCBs somewhere with the chips pre-installed. 

How would one do this?

nicolabortignon, 2023-12-22:
@the-baaron, @brushknight and I have recently started selling a [Smart Knob Dev Kit](https://www.seedlabs.it). It's a small batch of 300 for now as there is still quite a bit of manual work in the assembly process.
We wanted to get a Dev Kit out as soon as possible to the community, while we are developing more consumer focused products. The hardware has some changes from Scott's Smart Knob (ie.: Leds, Connectors, etc), but all the original features are included. The firmware is also different, but based on a fork from this codebase, so you should expect some level of similarities plus some nice new features like an Home Assistant integration. 

In the long run, we want to focus on controller / sensors / devices that help managing energy in your home. There are other initiatives that are exploring using haptic feedback as computer peripherals (i.e: Binaris).

the-baaron, 2023-12-28:
That sounds amazing! I'll be patient then, and sign me up as one of your first customers 🔥

tmarkson, 2024-01-02:
+1 to integration with Hass or Esphome

ChristiaanSl, 2024-01-02:
Awesome! I missed the first batch, but am very excited for a possible second batch. How can I get notified when it's in stock again? I don't wanna miss the next batch.

nicolabortignon, 2024-01-03:
I'll try not to spam this issue with a project that is parallel to this codebase. So I'll use this message to post more updates.

@tmarkson we have implemented mqtt as a protocol, so it should interface well with both HASS and ESPHome. We are testing mainly on HASS.
@ChristiaanSl on the product page there is a form, just fill that out and you'll get an email once the next batch is available

Frederik-VP, 2024-01-22:
> I'll try not to spam this issue with a project that is parallel to this codebase. So I'll use this message to post more updates.
>
> @tmarkson we have implemented mqtt as a protocol, so it should [...]

Could you please let me know how to use the mqtt implementation? I have my own mqtt server I like to query

Xayer, 2024-09-28:
I just want people to know that there's still a demand for this devkit. We would still love to buy them!

dannymcbroom, 2025-02-19:
Would JLCPCB be able to produce the assembled board?  I have tried to get this through their website.  I found the BOM file but I cant find the Pick and Place CPL file.  Does anyone know where that would be?