scottbez1/smartknob #160: upload firmware - task failed

pippo93, 2024-08-23:
Hi all,
I was trying to follow these instructions (https://paper.dropbox.com/doc/SmartKnob-firmware-and-software-VUb9nq7btuhnHw5KlgJIH) to configure the new arrived SmartKnob Dev Kit 0.1 from SeedLabs. 
Unfortunately, the process fail with this error:
`A fatal error occurred: This chip is ESP32-S3(beta3) not ESP32. Wrong --chip argument?`

Any suggestions?

frank-f, 2024-08-24:
Try using a lower baudrate. I've had to lower it to 460800 as well.

pippo93, 2024-09-03:
> Try using a lower baudrate. I've had to lower it to 460800 as well.

Tried to lower both “monitor_speed” and “upload_speed” in platformio.ini file as suggested but nothing changes.

frank-f, 2024-09-03:
Maybe double check all solder connections? Did you solder the components by hand or with stencil and reflow/hotplate? Try another CH340, if you have a spare?

pippo93, 2024-09-03:
> Maybe double check all solder connections? Did you solder the components by hand or with stencil and reflow/hotplate? Try another CH340, if you have a spare? Also, if you change the `monitor_speed` setting, make sure to also update `build_flags` accordingly, otherwise you're not going to see anything useful once you got the firmware to upload.

I boughy the assembled kit by SeedLabs. Today after one more try uploading the standard firmware as requested in the guide, the device definitely shut off (led and screen) and is no more recognized from the pc. I’ve asked to SeedLabs what I have to do because the device has now less then 15 minutes of life…

pippo93, 2024-09-07:
Faulty cable, sorry

closed 2024-09-07 by pippo93