scottbez1/smartknob #163: Can't flash firmware after manual boot mode

Oliver1948, 2024-10-06:
My base board has been completely soldered (minus the screen. I have tried many times to flash the firmware from vscode but it is not working. It says the following: Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (3.4.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6)
 - tool-esptoolpy @ 1.30100.210531 (3.1.0)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
[nanopb] Installing Protocol Buffers dependencies
Requirement already satisfied: protobuf>=3.19.1 in c:\users\oaruf\.platformio\penv\lib\site-packages (5.28.2)
[nanopb] No generation needed.
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TLV493D-Magnetic-Sensor @ 1.0.3
|-- AceButton @ 1.9.1
|-- PacketSerial @ 1.4.0
|-- Nanopb @ 0.4.7
|-- Simple FOC @ 2.2.0
|-- TFT_eSPI @ 2.4.25
|-- FastLED @ 3.5.0
|-- HX711 @ 0.7.5
|-- Adafruit VEML7700 Library @ 1.1.1
|-- FFat @ 1.0
Building in release mode
Retrieving maximum program size .pio\build\view\firmware.elf
Checking size .pio\build\view\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   6.6% (used 21716 bytes from 327680 bytes)
Flash: [====      ]  36.1% (used 473462 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM7
Uploading .pio\build\view\firmware.bin
esptool.py v3.1
Serial port COM7
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2
=============================================================== [FAILED] Took 29.85 seconds ===============================================================

Environment    Status    Duration
-------------  --------  ------------
view           FAILED    00:00:29.853
========================================================== 1 failed, 0 succeeded in 00:00:29.853 ========================================================== 

 *  The terminal process "C:\Users\oaruf\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--target', 'monitor'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

I saw a previous issue quite similar to mine, except for the fact that I have the transistors installed, and when I tried to manually enter the board into boot mode, nothing happened. In addition to all this, no lights turn on when I plug the board in, and the boot and en pins oscilate between 0 and 3.3v when attempting to connect. Below is a picture of my board for reference.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ChristianEmeis, 2024-10-07:
You need to use either U6 or U9, but not both. Maybe try to unsolder one of them and try again. If this does not help, try another USB-Cable, often upload errors are caused by faulty cables.

Oliver1948, 2024-10-07:
I have done both (new cable and removed U9), still nothing.

kawashimaken, 2024-12-14:
ch340c might be damaged, repace it with a new one, it may solve the problem.

kawashimaken, 2024-12-14:
> I have done both (new cable and removed U9), still nothing.
ch340c might be damaged, repace it with a new one, it may solve the problem.

Oliver1948, 2024-12-31:
Here's some more background info I should've put in my original issue. This board was ordered pcba from jlcpcb baring the microscopic motor chip and the main chip. Those are the two components I soldered on, and the board is not working. I am apprehensive to try and swap components that jlc installed cause I believe they tested everything before they sent it (doesn't mean I didn't fry something though 😂). Should I still swap ch340c?

Oliver1948, 2024-12-31:
> > I have done both (new cable and removed U9), still nothing.
> 
> 
> 
> ch340c might be damaged, repace it with a new one, it may solve the problem.
> 
> 

Here's some more background info I should've put in my original issue. This board was ordered pcba from jlcpcb baring the microscopic motor chip and the main chip. Those are the two components I soldered on, and the board is not working. I am apprehensive to try and swap components that jlc installed cause I believe they tested everything before they sent it (doesn't mean I didn't fry something though 😂). Should I still swap ch340c, or should I focus on the things I installed?

Oliver1948, 2025-01-02:
Just figured out my issue and finally got it all working! I had to change the upload speed to 115200, and I had to change the board to pico32. This was all figured out after ChatGPT helped me finally get a terminal connected to the esp32 chip so I could pull info from it. So yeah, AI is taking over. Thanks to everyone for the help!

closed 2025-01-02 by Oliver1948