scottbez1/smartknob #166: Llilygo's t-knob

TonyVie, 2024-12-20:
Hello !

Has anyone checked lilygo's t-knob ? 
https://lilygo.cc/products/t-knob

How does it compares to smartknob ?

tomByrer, 2024-12-28:
I think you meant this model?
https://lilygo.cc/products/t-encoder-plus?variant=44242407325877

nspitko, 2024-12-31:
The t-encoder is just a rotary encoder. t-knob uses a stepper motor and hall sensor like the smart knob. It doesn't have an LCD screen out of the box but the pass through provides spi as well as two generic IO pins, so it wouldn't be hard to add one.

Example code is here, https://github.com/Xinyuan-LilyGO/T-Knob

Similarities: Same motor driver and hall sensor
Differences: uses an esp32-C6-Mini (4mb flash), and only 4 LEDs. No strain gauge or LCD, though adding an LCD should be easy.

Should be possible to get a cut down version of the FW running on these devices, though we'll need to figure out a solution for the lack of a built-in "press" functionality.

stefan-sherwood, 2025-02-10:
One of those models has no screen. The other has no gimbal motor. The one with the screen substitutes the "press" functionality with a touchscreen, which opens up some interesting options.