scottbez1/smartknob #172: Is this repository still being maintained?

memphis242, 2025-05-13:
I see issues popping up and PRs that have been open for a while. I was considering contributing myself, but without active admins / maintainers, there's no capability to contribute :/.

jotalevi, 2025-05-19:
repo is not dead but it has little interaction lately, I'm sure there are some active admins and @scottbez1 tends to answer to stuff himself every once in a while. Also even if no one answers your PRs you can still contribute by forking the project.

memphis242, 2025-05-19:
To contribute through a fork, I'd need to merge back in here, right? Otherwise, the fork is just in my own little world, with little motivation for other people to use it if I'm only making minor tweaks.

voxpelli, 2025-05-19:
@memphis242 It has to start somewhere, an alternative is to find someone else's fork that look active and try to contribute to that one, to create some common ground and a higher likelihood for momentum

jotalevi, 2025-05-19:
Just like @voxpelli said, looking for some more active forks is a great way to get started. Here are some of the most recent forks:

[chrisjameschamp/smartknob](https://github.com/chrisjameschamp/smartknob)
[ryangeorge285/scrollknob](https://github.com/ryangeorge285/scrollknob)
[shuidobro/smartknob](https://github.com/shuidobro/smartknob)
[CC-Ian/smartknob](https://github.com/CC-Ian/smartknob)

jotalevi, 2025-05-19:
sure mate, that's the spirit. 
Just remember to close the issue once you find a nice fork to work on 
Take care!